platformio/platform-espressif32 #593: Can't use custom framework-espidf since v3.3.0

rohansingh, 2021-08-02:
Trying to specify a custom `framework-espidf` is broken since v3.3.0 with this error:

```
TypeError: write() argument must be str, not None:
  File "/usr/local/Cellar/platformio/5.1.1/libexec/lib/python3.9/site-packages/platformio/builder/main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/rohan/.platformio/platforms/espressif32/builder/main.py", line 218:
    target_elf = env.BuildProgram()
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/5.1.1/libexec/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/5.1.1/libexec/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/5.1.1/libexec/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 654:
    return method(*args, **kw)
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/rohan/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/rohan/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1052:
    create_version_file()
  File "/Users/rohan/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 841:
    fp.write(
```

I believe this is a regression caused by 4cb06337d678312dbcb293f508d7de7d7124ee68 (/cc @valeros), which was meant to resolve #558. Here is the code in question:

```py
def create_version_file():
    version_file = os.path.join(FRAMEWORK_DIR, "version.txt")
    if not os.path.isfile(version_file):
        with open(version_file, "w") as fp:
            fp.write(platform.get_package_version("framework-espidf"))
            fp.write(
                get_original_version(platform.get_package_version("framework-espidf"))
            )
```

For example, let's say I want to explicitly specify ESP-IDF v4.3 from the Espressif origin:

```ini
platform_packages =
  framework-espidf @ https://github.com/espressif/esp-idf#v4.3
```

In this case, `platform.get_package_version("framework-espidf")` will return `'1.1.0+sha.c9646ff'`. In that string, c9646ff is the Git commit hash for v4.3.

However, this causes `get_original_version` to return `None`:

```
def get_original_version(version):
    if version.count(".") != 2:
        return None
    _, raw = version.split(".")[:2]
    if int(raw) <= 99:
        return None
    if int(raw) <= 9999:
```

rohansingh, 2021-08-02:
Hmm yeah, looking into this and thinking about it a bit more, I'm not sure how to proceed. ESP-IDF uses `git describe` in this situation to figure out the most recent tagged version in the tree. But we can't do that.

Perhaps we could add `board.esp-idf.IDF_VER` and prefer that if it's available?

valeros, 2021-08-17:
Hi @rohansingh ! Thanks for reporting, should be fixed in the development branch. Please retest with the upstream version of the platform:
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
...
```

closed 2021-08-17 by valeros

rohansingh, 2021-08-17:
Oh awesome, looks great. Thanks!

dumarjo, 2021-11-09:
Any workaround to be able to use  the espidf and arduino ?

mvdberge, 2021-11-24:
> @klvnptr That's a separate issue, wherein you can't actually use an ESP-IDF branch directly. Instead you have to make a fork that contains a `package.json`.
> 
> That's always been the case, though it's not really documented anywhere.

Do you have a working example somewhere which could be used for reference? I'm new to platform.io and this would help a lot.
Thanks.

commit 5001ac45 (2021-08-17): Properly handle ESP-IDF package version for custom packages

Resolve #593