platformio/platform-espressif32 #597: ESPRandom library won't compile in 2.0.0-rc1, std::vector error

haddow777, 2021-08-09:
Hello, I figured I would check here on this issue, as the ESPRandom library compiles in 1.0.6 fine, but fails with a load of errors when I'm using 2.0.0-rc1. I don't get it either. It says it doesn't like std::vector, which doesn't make sense to me. Anyways, here's some of the output I'm getting

`
Compiling .pio/build/nodemcu-32s/liba5e/ESPRandom/ESPRandom.cpp.o
Compiling .pio/build/nodemcu-32s/lib827/BLE/BLE2902.cpp.o
Compiling .pio/build/nodemcu-32s/lib827/BLE/BLE2904.cpp.o
In file included from .pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.cpp:1:
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:56:15: error: 'vector' in namespace 'std' does not name a template type
   static std::vector<uint8_t> uuid();
               ^~~~~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:56:10: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:12:1:
+#include <vector>
 
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:56:10:
   static std::vector<uint8_t> uuid();
          ^~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:63:15: error: 'vector' in namespace 'std' does not name a template type
   static std::vector<uint8_t> uuid1();
               ^~~~~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:63:10: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
   static std::vector<uint8_t> uuid1();
          ^~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:70:15: error: 'vector' in namespace 'std' does not name a template type
   static std::vector<uint8_t> uuid4();
               ^~~~~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:70:10: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
   static std::vector<uint8_t> uuid4();
          ^~~
.pio/libdeps/nodemcu-32s/ESPRandom/ESPRandom.h:87:41: error: 'vector' in namespace 'std' does not name a template type
`

It keeps on going like that for a while. I haven't used vectors in my code as I like to avoid them unless totally necessary. Anyways, I'll put my platform.ini file in here too:

`
[env:nodemcu-32s]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = nodemcu-32s
framework = arduino
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0-rc1
upload_port = /dev/cu.SLAB_USBtoUART
upload_speed = 921600
monitor_speed = 115200
lib_deps = 
	gilmaimon/ArduinoWebsockets@^0.5.2
	bblanchon/ArduinoJson@^6.18.3
	protohaus/ESPRandom@^1.4.1
board_build.partitions = partitions_large.csv
build_type = debug
monitor_filters = log2file, default, esp32_exception_decoder
`

So, I don't even have the code included. I initially installed it and it failed. In my process of trying to figure out what is going wrong, I backtracked the arduino-esp32 platform back to stable just to make sure it wasn't the issue. Then the library compiled fine. So I brought it back to 2.0.0-rc1 and it failed the same way again. Please let me know if I'm missing something or if this issue may be getting caused by something other than arduino-esp32. Thank you.

valeros, 2021-08-09:
Hi @haddow777 ! Is there any chance you could check that a similar project works just fine in the Arduino IDE? This way we will be sure there is no problem with the framework itself.

haddow777, 2021-08-10:
I am trying. This doesn't seem to be my time for success. I have been using ArduinoIDE alongside PlatformIO for small things. I've found testing some things there in small dedicated programs easier as when something goes wrong, you know something else isn't contributing to it.
Unfortunately, something has changed since the last time I tried to use it. Now the whole thing is messed up. I can't even compile a simple Hello World output to the Serial at the moment. I am looking into it, but finding that the error message seems to be generic as articles with it go back years. No real solutions offered though, so I can't say when I will be able to check this out. Here is the error message ArduinoIDE is giving me:

```
[9052] Error loading Python lib '/var/folders/wm/36r934dn0hqf8tgsvqrsclsh0000gp/T/_MEIgKBHUo/libpython3.8.dylib': dlopen: dlopen(/var/folders/wm/36r934dn0hqf8tgsvqrsclsh0000gp/T/_MEIgKBHUo/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/wm/36r934dn0hqf8tgsvqrsclsh0000gp/T/_MEIgKBHUo/libintl.8.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/wm/36r934dn0hqf8tgsvqrsclsh0000gp/T/_MEIgKBHUo/libintl.8.dylib
exit status 255
```

haddow777, 2021-08-10:
Okay, I still haven't found a fix for my ArduinoIDE issue. On a somewhat more positive note, i guess I don't need to. Long story short, I added the library anyways. I somehow got the idea that this failure I am having is a post compile issue. I'm looking for compiler errors, so I should still get errors if there are any. I installed the library. I should note it's not included in the Arduino IDE library, I installed it by zip. 
Ultimately, it died with the same errors. There's a tonne of includes in Arduino.h and I would have assumed that std::vector would have been one of them. I guess not. So it looks like the library is relying on an old expectation that no longer exists. I'm not an expert, but I think that means the issue is with ESPRandom being out of date and not platform-espressif32's issue. I'm going to go through this issue on their board.
As a side point, do you know of another library that may be up to date that provides uuid's? That's all I really wanted from the library. Otherwise I may just manually include the library code and try fixing it myself.
Thank you.

haddow777, 2021-08-10:
Hi @valeros, never mind that last statement (I was going to delete it, but thought that may be more confusing if you already read it). I'm really sorry about this. I'm certain I regressed arduino-esp32 in platformIO before and it came up with the same compile errors. I went a little further in ArduinoIDE and regressed it back to arduino-ess32 1.0.6. It compiled fine. I then came back to PlatformIO and compiled it under 1.0.6 and 2.0.0-rc1 again (at least I thought so, somehow I must have messed that up before). It now compiles fine under 1.0.6 and not under 2.0.0-rc1.

It seems that std::vector is normally included in the main arduino library in the earlier version, but isn't included in the latest version. Frankly, I don't know where that issue lies. For all I know, the arduino-esp32 maintainers may have decided not to include it in the latest version. In any event, there is the latest I know on the issue.

valeros, 2021-08-10:
Thanks for looking into the problem, it seems that issue has nothing to do with PlatformIO. Anyway, you can always lock your platform version so your project won't be affected by the new updates of the Arduino core:
```ini
[env:nodemcu-32s]
platform = espressif32@~3.3.0
board = nodemcu-32s
framework = arduino
```

closed 2021-08-10 by valeros