platformio/platform-espressif32 #599: Upgrading Espressif 32 from 3.3.0 to 3.3.1 broke everything

GeorgeFlorian, 2021-08-09:
### Configuration

**Operating system**: Linux Mint

**PlatformIO Version** (`platformio --version`):
```
Updating platformio/contrib-piohome           3.3.4 @ ~3.3.4                     [Up-to-date]
Updating platformio/contrib-pysite            2.38.191020 @ ~2.38.0              [Incompatible 2.39.201019]
Updating platformio/tool-scons                4.40100.2 @ ~4.40100.2             [Incompatible 4.40200.0]

Platform Manager
================
Platform espressif32
--------
Updating platformio/espressif32               3.3.1                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10006.210326 @ ~3.10006.0        [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-openocd-esp32        2.1000.20210721 @ ~2.1000.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
```

### Description of problem

Upgrading Espressif 32 from 3.3.0 to 3.3.1 broke everything.

I have errors inside libraries where I did not have before.
**platformio.ini** modified itself after the update.
I have no idea why it says **Incompatible** above.
Not even the simplest Blink sketch works.

#### Steps to Reproduce

1. Open Blink project from Project examples
2. Upload sketch to any ESP32
3. Watch Serial Monitor

### Actual Results

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0018,len:4
load:0x860020c0,len:2
ho 12 tail 2 room 2
load:0x941c0422,len:537354562
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0018,len:4
load:0x860020c0,len:2
ho 12 tail 2 room 2
load:0x941c0422,len:537354562
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
ets Jun  8 2016 00:22:57
```

### Expected Results

Whatever the blink project should output.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32dev
framework = arduino
board_build.flash_mode = qio
#LINUX
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
#WINDOWS:
#upload_port = COM
#monitor_port = COM

monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
```

**Source file to reproduce issue:**
```cpp
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>

// Set LED_BUILTIN if it is not defined by Arduino framework
#define LED_BUILTIN 22

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}
```

### Additional info

I was making some finishes touches on a project when I decided to upgrade from Espressif 32 3.3.0 to 3.3.1 and nothing worked since.

Also, on that project, the **platformio.ini** file modified itself after the update.
FROM:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32dev
framework = arduino
board_build.flash_mode = qio
#LINUX
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
#WINDOWS:
#upload_port = COM
#monitor_port = COM

monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5

#libraries
lib_ldf_mode=deep
lib_deps =  https://github.com/me-no-dev/ESPAsyncWebServer.git
            https://github.com/me-no-dev/AsyncTCP.git
            https://github.com/marcmerlin/SmartMatrix_GFX.git
            https://github.com/marcmerlin/Framebuffer_GFX.git
            https://github.com/adafruit/Adafruit-GFX-Library.git
            https://github.com/FastLED/FastLED.git
```
TO:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32dev
framework = arduino
board_build.flash_mode = qio
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
lib_deps = 
	https://github.com/me-no-dev/ESPAsyncWebServer.git
	https://github.com/me-no-dev/AsyncTCP.git
	https://github.com/marcmerlin/Framebuffer_GFX.git
	https://github.com/marcmerlin/SmartMatrix_GFX.git
	https://github.com/adafruit/Adafruit_BusIO.git
	https://github.com/adafruit/Adafruit-GFX-Library.git
	https://github.com/FastLED/FastLED.git
	Wire
```

valeros, 2021-08-09:
Hi @GeorgeFlorian ! 

> Upgrading Espressif 32 from 3.3.0 to 3.3.1 broke everything.

FYI, you can always roll back to the previous version of the platform in your `platformio.ini`:
```ini
[env:esp32doit-devkit-v1]
platform = espressif32@3.3.0
board = esp32dev
framework = arduino
...
```

> I have errors inside libraries where I did not have before.

What were the errors exactly?

> platformio.ini modified itself after the update.

PlatformIO doesn't touch your configuration files unless you're editing them using the `Projects` tab on the `PlatformIO Home` page.

> I have no idea why it says Incompatible above.

The PlatformIO build system is sensitive to the version of Python interpreter you're using. For some reason several internal packages were installed for Python 3.9, but from the logs above you're trying to run PlatformIO using Python 3.8. If this what you want, then try to run `pio update` in your terminal to pull proper packages.

> Not even the simplest Blink sketch works.

Looking at your configuration the only potential change that might have caused the issue comes from https://github.com/platformio/platform-espressif32/issues/584 . The flash mode is now set according to the value in the board manifest, so please make sure the new value is good for your board.

GeorgeFlorian, 2021-08-09:
> Hi @GeorgeFlorian !
> FYI, you can always roll back to the previous version of the platform in your `platformio.ini`:
> 
> ```ini
> [env:esp32doit-devkit-v1]
> platform = espressif32@3.3.0
> board = esp32dev
> framework = arduino
> ...
> ```
I just did this: `platform = espressif32@3.3.0` and everything works just fine
 
> What were the errors exactly?

I did not document those, because I am not sure if these are related.
I have a strange behavior with Platformio where some libraries that are linked in `platformio.ini` get downloaded twice:
[image]
```
#libraries
lib_ldf_mode=deep
lib_deps =  https://github.com/me-no-dev/ESPAsyncWebServer.git
            https://github.com/me-no-dev/AsyncTCP.git
            https://github.com/adafruit/Adafruit-GFX-Library.git
            https://github.com/marcmerlin/SmartMatrix_GFX.git
            https://github.com/marcmerlin/Framebuffer_GFX.git
            https://github.com/FastLED/FastLED.git
```
On the 3.3.1 version the libraries conflicted with each other.

> PlatformIO doesn't touch your configuration files unless you're editing them using the `Projects` tab on the `PlatformIO Home` page.

I think this happened when I was trying to get rid of the doubled libraries by uninstalling them from `Platformio > Libraries > Installed`

> Looking at your configuration the only potential change that might have caused the issue comes from #584 . The flash mode is now set according to the value in the board manifest, so please make sure the new value is good for your board.

I have used the same configuration for the past year or so and I have never had troubles with it:
```
[env:esp32doit-devkit-v1]
platform = espressif32@3.3.0
board = esp32dev
framework = arduino
board_build.flash_mode = qio
```
I am using a DevKitV4 with a Wrover-B on it.
[image]

> The PlatformIO build system is sensitive to the version of Python interpreter you're using. For some reason several internal packages were installed for Python 3.9, but from the logs above you're [...]

This is oblivious to me.
The Python3 version that I have installed on the PC is `Python 3.8.10`.
`pio` / `platformio` commands do not work inside VSC terminal for me.

valeros, 2021-08-09:
> I did not document those, because I am not sure if these are related.

It's quite hard to guess whether those errors are related without the additional information, but I suppose you managed to get rid of them?

> I have a strange behavior with Platformio where some libraries that are linked in platformio.ini get downloaded twice:

You specified your library dependencies as github repos without any specific labels, so when there are new commits in those repos PlatformIO will pull that changes automatically (and back up previous versions). I'd recommend you to remove the entire `.pio` folder and try to compile your project again.

> I think this happened when I was trying to get rid of the doubled libraries by uninstalling them from Platformio > Libraries > Installed

That's true, PlatformIO modify your `platformio.ini` when you install/remove a library via PlatformIO Home and comments are not preserved, so this is expected.

> I have used the same configuration for the past year or so and I have never had troubles with it:

Any reason why you used `qio` as the flash mode? What happens if you specify `board_build.flash_mode = dio`?

> pio / platformio commands do not work inside VSC terminal for me.

It might be a good idea to delete the current PlatformIO installation and reinstall everything from scratch. Try deleting the folder `~/.platformio` on Unix or `%HOMEPATH%\.platformio` if you use Windows.

GeorgeFlorian, 2021-08-09:
> It's quite hard to guess whether those errors are related without the additional information, but I suppose you managed to get rid of them?

I have deleted the library links from `platformio.io`, copied every library from `.pio/libdeps` to `/lib` and deleted the doubles. Build the project again and this is how it looks:
[image]
As you can see, some of them got downloaded again and now there are 3 of them.
I have no idea how this works or how to fix it and use just one library.
I would really like some help here. It looks tragic.

> You specified your library dependencies as github repos without any specific labels, so when there are new commits in those repos PlatformIO will pull that changes automatically (and back up [...]

I have been doing this all day, they just get downloaded in doubles again.

> That's true, PlatformIO modify your platformio.ini when you install/remove a library via PlatformIO Home and comments are not preserved, so this is expected.

I had no idea this might happen. I was trying to get rid of the double libraries.

> Any reason why you used qio as the flash mode? What happens if you specify board_build.flash_mode = dio?

No idea. The boards work fine with or without this line.

> It might be a good idea to delete the current PlatformIO installation and reinstall everything from scratch. Try deleting the folder ~/.platformio on Unix

I did. The commands still don't work:
```
georgeflorian@georgeflorian-desktop:~/Documents/PlatformIO/Projects/ESP_TimerOnInput-main$ pio --version

Command 'pio' not found, but there are 16 similar ones.

georgeflorian@georgeflorian-desktop:~/Documents/PlatformIO/Projects/ESP_TimerOnInput-main$ platformio --version
platformio: command not found
georgeflorian@georgeflorian-desktop:~/Documents/PlatformIO/Projects/ESP_TimerOnInput-main$ 
```

valeros, 2021-08-10:
> I have deleted the library links from platformio.io, copied every library from .pio/libdeps to /lib and deleted the doubles.

Please don't do that, just delete the `.pio` folder and try to compile your project again, PlatformIO will download all required libraries (including dependencies) specified in your `platformio.ini`. 

> I don't remember why I've been using board_build.flash_mode = qio, but removing it makes 3.3.1 work.

Good, let's stick to the configuration without this line.

> I did. The commands still don't work:

How you're compiling your project then? In first post you even specified the output from the `platformio --version` command.  Can you run `platformio system info` and attach here the output?

RoscoeTheDog, 2021-08-19:
Bump. I am having this same issue. I updated to see if there was a bugfix for a well-known Wifi bug and it broke my _entire_ vscode/platformio environment. I cannot compile a project that was working before the update to 3.3.1 I get the following error:

```C:\Users\admin\Documents\Github\C++\ESP32-S2-Saola\ESP32_S2_Saola1/src/rtosTasks.c:728: undefined reference to `vTaskList'```

Note this is my own project source file, but `vTaskList()` is a valid espidf method part of the freeRTOS framework which compiled before and now it seems the linker can't find it. Also it struggles to link or find modules like the power management module, <esp_pm.h>.

What is worse is that I cannot revert the changes! I've deleted my pio folder, cleaned the project, restarted VSCode, even manually uninstalled all espidf packages through the platformIO menu but everytime I build I come up with the same error. Below is my platformio.ini config. You can see I am declaring and specifying 3.3.0 but I am still getting the same errors and stuff from 3.3.1.

This is a huge inconvenience and has halted all my development when I actually have time to code and work on stuff. What is a guaranteed workaround or solution to getting the old version of the framework working? At this rate, I might have to restore to a snapshot in my virtual machine and re-pull from github just to get my development environment back. Seems crazy.

valeros, 2021-08-19:
Hi @RoscoeTheDog ! Could you please put together a a minimal example to reproduce the issue? Also, you sure you were using v3.3.0 before the update to 3.3.1?

RoscoeTheDog, 2021-08-19:
@valeros It is entirely possible that 3.3.0 is newer than what I'm working with currently and also has problems building. I restored a snapshot of my vm and when checking the versions during a build, it seems I was using a version older than that.

[image]

valeros, 2021-08-19:
According to the screenshot you used ESP-IDF v4.2.1, so I suppose rolling back to `platform = espressif32@3.2.1` should revive your project. The latest versions of the platform `>= 3.3.0` use ESP-IDF v4.3, so you need to ensure that your project is compatible with that version of ESP-IDF.

RoscoeTheDog, 2021-08-19:
> so you need to ensure that your project is compatible with that version of ESP-IDF.

I just checked the espidf latest documentation to see if the modules and methods in the older versions of IDF I was using prior are still used in the new one and they are. So why would they fail to build? It seems like there is likely something else going on here unrelated to IDF compatibility unless I am missing something

valeros, 2021-08-20:
@RoscoeTheDog Could you please put together a minimal example to reproduce the issue?

handmade0octopus, 2021-09-05:
Just FYI after update I have similar issue:

```rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57```

After changing to 

```
    "flash_mode": "qout",
    ```
    
    From qio it works but its very confusing and why does it do that?

valeros, 2022-05-04:
Feel free to reopen the issue if the problem persists with the latest version of the `espressif32` dev-platform.

closed 2022-05-04 by valeros