platformio/platform-espressif32 #604: Can't embed files having spaces in name

ssilverman, 2021-08-14:
What kind of issue is this?

- Selected: **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS v11.5.2

**PlatformIO Version** (`platformio --version`): 5.1.1

### Description of problem

`board_build.embed_txtfiles` doesn't seem to allow files having spaces in their names. Even quoting the name doesn't work. If I include a file having the name "First Name.txt", the error looks like this:
```
*** [.pio/build/esp32dev/"First.txt.o] Source `"First' not found, needed by target `.pio/build/esp32dev/"First.txt.o'.
```

#### Steps to Reproduce

1. Create a simple project (I used "esp32dev") having a file named "First Name.txt".
2. Add this to the config: `board_build.embed_txtfiles = "First Name.txt"`
3. Try to build.

### Actual Results

I see an error that looks like the above.

### Expected Results

The build succeeds.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_build.embed_txtfiles = First Name.txt
```

**Source file to reproduce issue:**
```c++
#include <Arduino.h>

extern const uint8_t kFile[] asm("_binary_First_Name_txt_start");
extern const uint8_t kFile_end[] asm("_binary_First_Name_txt_end");
const int kFile_len = kFile_end - kFile - 1;

void setup() {
  log_printf("len = %d\n", kFile_len);
}

void loop() {
}
```

### Additional info

The build succeeds if I change the included file to "FirstName.txt". However, I find it odd that using double quotes around "FirstName.txt" in the ini file causes the builder to use double quotes in the name, thus failing the build.

valeros, 2021-08-16:
Hi @ssilverman ! Many thanks for reporting, should be fixed in the dev branch. Please retest your project with the upstream version of the platform (a [Git Client](https://git-scm.com/downloads) is requried):
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
monitor_speed = 115200
; Double quotes are not requried
board_build.embed_txtfiles = First Name.txt
```

closed 2021-08-16 by valeros

ssilverman, 2021-08-17:
Thank you for putting this in the correct spot, I wasn't quite sure which section it needed to go in. :)

I tried the fix, changing `platform` to the above, but still saw the error until I did "Update All". Now it appears to work.

ssilverman, 2021-08-17:
Side note: The build still disallows a file name surrounded by double quotes. Here is the error:
```
*** [.pio/build/esp32dev/"First Name.txt".txt.o] Source `"First Name.txt"' not found, needed by target `.pio/build/esp32dev/"First Name.txt".txt.o'.
```

valeros, 2021-08-17:
That's actually expected, there is no need to wrap file names in double quotes as spaces in names are handled internally.

commit bec7adea (2021-08-16): Properly handle file names specified as binary data

Resolve #604