platformio/platform-espressif32 #616: v3.3.2 Does not compile with Arduino

calcut, 2021-09-03:
Apologies if this is a duplicate, it seems closely related to:
https://github.com/espressif/arduino-esp32/issues/5436

However, I had hoped that the new v3.3.2 release might have resolved this.
When trying to compile a blink example, I get this error

```
Linking .pio/build/esp32-s2/firmware.elf
/Users/calum/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32-s2/firmware.elf] Error 1
```

It seems OK when I use platform-espressif32.git#feature/arduino-upstream
Is that expected still? 

[env:esp32-s2]
; platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform = https://github.com/platformio/platform-espressif32#v3.3.2
board = esp32dev
framework = arduino
board_build.mcu = esp32s2

platform_packages = 
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0

upload_port = /dev/cu.usbmodem01

valeros, 2021-09-03:
v3.3.2 was a maintenance release and didn't include the latest Arduino core.  Anyway, we're working on integrating Arduino core v2.0, so it should be fixed in the dev branch  pretty soon.

Jason2866, 2021-09-03:
Do you use esptool.py v.3.1? The new toolchains do need this as minimal version.
I have the error you describe from time to time too. I dont know the real reason but deleting the hidden folder `.platformio` and if you use the cache function of Platformio delete the folder `.cache` too. After doing this compiling does always worked again (for me)

calcut, 2021-09-03:
Yes, using esptool 3.1
I'll try your suggestion next time I'm using platformIO, but I have reverted to ArduinoIDE due to this other issue https://github.com/platformio/platform-espressif32/issues/591

Miq1, 2021-09-28:
> v3.3.2 was a maintenance release and didn't include the latest Arduino core. Anyway, we're working on integrating Arduino core v2.0, so it should be fixed in the dev branch pretty soon.

This statement was a bit bold, so it seems. Are there any predictions as when to expext the 2.0 core integration?

viniciusdesa, 2021-12-02:
Looking forward for this integration

valeros, 2022-04-18:
Should be resolved in https://github.com/platformio/platform-espressif32/commit/581c7d03252f19f5a029f190ff4b2d1b62db157f

closed 2022-04-18 by valeros