platformio/platform-espressif32 #618: xtensa-esp32-elf-g++: error: esp32dev.map: No such file or directory *** [.pio/build/esp32dev/firmware.elf] Error 1

jitendra1607, 2021-09-08:
https://github.com/espressif/arduino-esp32/issues/5646

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages =
   framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0
board_build.filesystem = littlefs
```

xtensa-esp32-elf-g++: error: esp32dev.map: No such file or directory
*** [.pio/build/esp32dev/firmware.elf] Error 1

with the above platformio.ini, I got above error on build. If i remove below line then no issues.
```
platform_packages =
   framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0
```

Platform - macos
board- esp-wroom-32

Jason2866, 2021-09-08:
You have too change:
```
platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
```
As long tnere is no officiall release from platformio for core 2.0.

Jason2866, 2021-09-09:
Will be solved with #619

valeros, 2022-05-04:
Resolved in the latest Arduino package.

closed 2022-05-04 by valeros