platformio/platform-espressif32 #619: Support for the latest Arduino v2.0

valeros, 2021-09-09:
The latest Arduino core adds support for ESP32S2, ESP32C3 SoCs and a great amount of new boards. Related issues:
- https://github.com/platformio/platform-espressif32/issues/591
- https://github.com/platformio/platform-espressif32/issues/557
- https://github.com/platformio/platform-espressif32/issues/564
- https://github.com/platformio/platform-espressif32/issues/570
- https://github.com/platformio/platform-espressif32/issues/546
- https://github.com/platformio/platform-espressif32/issues/521
- https://github.com/platformio/platform-espressif32/issues/609
- https://github.com/espressif/arduino-esp32/issues/5255

MichaelDu9226, 2021-09-13:
Hi all
Are there any plans for this feature? When will it be supported?

maxgerhardt, 2021-09-14:
You might also want to take a look at https://github.com/espressif/arduino-esp32/issues/5505 in regards to this.

forkineye, 2021-10-14:
> I've been trying to understand how I can get Arduino v2.0.0 to work in PlatformIO because there are some important bugfixes in esp-idf v4.3 which I need.. I have tried building with Arduino as an IDF component but that only supports up to IDF v4.0 and is buggy. I can't think of a way to get these updates without a lot of rework on my project by converting entirely to IDF.
> 
> Maybe it is my lack of experience with the platform, but I am confused about what prevents us using Arduino v2.0 in PlatformIO, can you explain it for me?

Here's an example that may help you out.  The relevant tidbits are in [esp32git] which just extends the [esp32] entry I have an overrides the platform until support is added.

```ini
;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;
; ESP32 defaults for 4MB flash                                     ;
; https://docs.platformio.org/en/latest/platforms/espressif32.html ;
;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;
[esp32]
platform = espressif32
board_build.filesystem = littlefs
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder
build_flags =
    ${env.build_flags}
lib_deps =
    ${env.lib_deps}
    esphome/AsyncTCP-esphome @ 1.2.2

;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;
; ESP32 pulling from upstream core ;
;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;
[esp32git]
extends = esp32
platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0

;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;
; Build targets (environments) ;
;~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~;

; Lolin D32 Pro
[env:d32_pro]
extends = esp32git ; use until 2.0.0 core makes it into PlatformIO
board = lolin_d32_pro
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -mfix-esp32-psram-cache-strategy=memw
```

olicooper, 2021-10-15:
Thanks for your response @forkineye !
I followed your advice and created a project with your suggestions but I am unable to get a simple project to build still in PlatformIO. Although it is compiling, it is not linking. I Can't see anything different in my configuration that would cause this.
The project can be found here: https://github.com/olicooper/idf43-arduino2-test

louislewis2, 2021-10-19:
While I am aware there are a few workarounds to get going with v2 as shown above.
It would be great to get a rough idea when this work planned for. 

Thanks

tablatronix, 2021-11-10:
I have been playing with this for a few days with an esp32 S2, I cannot get OTA to work at all, updater class always aborts on erasepartition.. Its like there is something wrong with my partitions, if i compile with arduino ide, it works fine.

There also seems to be some timing issues with wifi funcs, adding typical workarounds like delays after setmode etc seems to be fixing most of them..

Anyone have OTA working  with #arduino-upstream

Jason2866, 2021-11-12:
@tablatronix  yes, Tasmota is doing OTA with.
Install via https://tasmota.github.io/install/
Connect to AP 192.168.1.4 do your wifi settings. Open the website of the device (via ip) and do a OTA upgrade (for example `http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin`) It just works. [Tasmota](https://github.com/arendst/Tasmota) is compiled with Platformio

tablatronix, 2021-11-12:
hmm I have not tried httpota, only esptool from ide

Jason2866, 2021-11-13:
Tasmota is compiled with a changed setup of sdkconfig in the Arduino Lib builder and has the same code base as release core 2.0.1

tablatronix, 2021-11-13:
It looks like it is using feature/arduino-idf-master, I cannot get it to compile for S2 though
`/Users/shawn/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value`

dexter323i, 2021-11-15:
Hi! I'd like to use an ESP32-S2-Saola-1 board for simulating different USB devices. No OTA, no SPIFFS, no any special needs. Just get a  gamepad arduino example code working. Could you guys help me out with a working platformio.ini config? I googled it, but found so many different settings from June until now. Suggesting different alpha and rc1 releases, different toolchains, commit hashes, etc. Which are all outdated by right now, I guess.
It looks like for some of you it is working somehow... Any ini config help would be appreciated!

Jason2866, 2021-11-16:
@dexter323i 
You can use this.
```
[env:esp32-s2]
framework = arduino
board = esp32-s2-saola-1
platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages           = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/2.0.1/framework-arduinoespressif32-release_IDF4.4.tar.gz
                              platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
```

visicon-masc, 2021-11-16:
To Make it work I had to:

- in the foillowing file i changed #if to #ifdef because of:
`.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\locale_facets_nonio.tcc:773:21: error: #if with no expression`

- and `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master`
results in: 
`Linking .pio\build\esp32dev_main_ref1\firmware.elf
c:/users/masc/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev_main_ref1\firmware.elf] Error 1`

Therefor I changed it to:
`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`

tablatronix, 2021-11-16:
I had same issue, ppl keep saying to use IDF branch, I cannot get it to compile, so not sure whats going on

Jason2866, 2021-11-16:
The setup i posted does work without any changes. If not delete the hidden folder `.platformio` You have left overs which results in the errors you had

visicon-masc, 2021-11-17:
I can confirm: `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master` works after deleting .platformio folder in user directory. (whilst it was a little fiddly to get platformio working again: had to reinstall platformio extention to get it straigt again, dunno why)

mhaberler, 2021-11-17:
> I can confirm: `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master` works after deleting .platformio folder in user directory. (whilst it was a little [...]

confirmed here as well

douardda, 2021-11-17:
I am also using esp32-arduino v2 (2.0.0) on [my version of the AR488](https://github.com/douardda/AR488/tree/esp32) GPIB interface  code and it works ok on the esp32s2-soala.

mloebl, 2021-11-19:
Thank you! Been chasing this for weeks now and using the old ` platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream` , deleting .platformio in my home folder fixed it here too.

tablatronix, 2021-11-20:
What do you mean delete .platformio ? That is my entire pio enviroment, is there a specific  package that needs rebuilding?

tablatronix, 2021-11-20:
Ok so I deleted my toolchains only and I think that fixed it, apparantly extensa esp32 and s2 toolchains were not updating even if specified in platform_pacakges

tablatronix, 2021-11-20:
well OTA almost works now.. sigh

```
Progress: 99%Progress: 99%Progress: 99%Progress: 99%[ 57853][D][Updater.cpp:211] _writeBuffer(): partitionEraseRange
[ 57853][D][Updater.cpp:213] _writeBuffer(): offset:651264
[ 57853][D][Updater.cpp:214] _writeBuffer(): SPI_FLASH_SEC_SIZE:4096
[ 57889][D][Updater.cpp:220] _writeBuffer(): partitionWrite
[ 57899][D][Updater.cpp:226] _writeBuffer(): ESP_IMAGE_HEADER_MAGIC
Progress: 99%[ 57905][D][Updater.cpp:211] _writeBuffer(): partitionEraseRange
[ 57905][D][Updater.cpp:213] _writeBuffer(): offset:655360
[ 57906][D][Updater.cpp:214] _writeBuffer(): SPI_FLASH_SEC_SIZE:4096
[ 57943][D][Updater.cpp:220] _writeBuffer(): partitionWrite
[ 57948][D][Updater.cpp:226] _writeBuffer(): ESP_IMAGE_HEADER_MAGIC
Progress: 100%Error[4]: End Failed
[ 57974][E][ArduinoOTA.cpp:357] _runUpdate(): Update ERROR: Flash Read Failed

```

mloebl, 2021-11-21:
> well OTA almost works now.. sigh
>
> ```
> Progress: 99%Progress: 99%Progress: 99%Progress: 99%[ 57853][D][Updater.cpp:211] _writeBuffer(): partitionEraseRange
> [ 57853][D][Updater.cpp:213] [...]

Same, I googled `Flash Read Failed` and didn't find a lot of hits. I posted yesterday the Espressif ESP32/Arduino forum but no response quite yet.

tablatronix, 2021-11-21:
Working now, I switched back to jasons exact ini and ota is working now..

mloebl, 2021-11-21:
I tried jasons' config, but still can't upload via ota unfortunately.

tablatronix, 2021-11-21:
Did you remove your toolchains and let it pull them again?

Jason2866, 2021-11-21:
Yes, if you remove the folder `.platformio` this is done. If a old toolchain is there it can happen that this one (not compatible) is used. You can try to remove stuff step by step until it is working successfully.

mloebl, 2021-11-21:
I've completely removed the .platformio folder, copied back in my couple libraries to /lib and same issue. Am I missing something obvous?
```
[env:esp32dev]
; platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
; platform = espressif32
platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages           = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/2.0.1/framework-arduinoespressif32-release_IDF4.4.tar.gz
                              platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
board = esp32dev
framework = arduino
board_build.mcu = esp32s2
monitor_speed = 115200
monitor_port = /dev/ttyUSB0
; upload_port = /dev/ttyUSB0
upload_port = 192.168.5.141
upload_protocol = espota
upload_flags = --host_port=50000
```

```
Uploading: [============================================================] 100% Done...

09:38:43 [INFO]: Waiting for result...
09:38:43 [INFO]: Result: Flash Read Failed

09:38:43 [INFO]: Result: 
09:38:43 [INFO]: Result: 
09:38:43 [INFO]: Result: 
09:38:43 [INFO]: Result: 
09:38:43 [ERROR]: Error response from device
*** [upload] Error 1
```

Thanks!

Jason2866, 2021-11-21:
@mloebl Does it work with Arduino IDE? If not this is a issue for ESP32 Arduino github.

KevinESP32, 2021-11-22:
Hello everybody, first time I am using github "forum",
I recently started a project with Arduino IDE on ESP32 DevKit, and moved into PlatformIO.
I experencied some bugs with the WiFi (using SoftAP) with Platform and then wanted to update the Arduino ESP Core 
to version 2.0 which was the one used by Arduino IDE.
I have a lot of troubles to makes it work, I have read a lot on chat about this and I am currently using this platform.ini file :

 **[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
board = esp32dev
framework = arduino
platform_packages =
		framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
upload_speed = 921600
board_build.f_cpu = 240000000L   ; Frequ CPU = 240MHz
board_build.f_flash = 80000000   ; Flash frequ = 80MHz
board_build.flash_mode = qio
board_build.partitions = huge_app.csv
monitor_speed = 115200**

It compiles, but after uploading to mcu, the ESP32 crash and try to reboot infinitely .....
Does anyone can give me help to fix this ?

In the terminal printing this in loop :
**rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0030,len:1240
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46**

Jason2866, 2021-11-22:
@KevinESP32 
Try `board_build.flash_mode = dio`

tablatronix, 2021-11-23:
Hi @KevinESP32  just to clarify this is bot a forum but for reporting/discussing issues with the library/platform

KevinESP32, 2021-11-23:
> @KevinESP32 Try `board_build.flash_mode = dio`

Thank you Jason2866, it was this ! How did you know about ?
Is QIO mode not implemented on ESP32 Dev, when compiling with Arduino IDE and with QIO mode selected, ESP32 is starting with DIO Flash mode, it seems it don't care about QIO

Jason2866, 2021-11-23:
@KevinESP32  See https://github.com/platformio/platform-espressif32/issues/584

KevinESP32, 2021-11-23:
Ok, so to be precise its works now with:
platform = **https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream**

**But there is the link error** using:  **https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master**
(I also removed the platformio folder before changing)

I am using in all the cases:
platform_packages =
               framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
		toolchain-xtensa-esp32 @ 8.4.0+2021r1
		toolchain-xtensa32@~2.80400.0
		platformio/tool-esptoolpy @ ~1.30100

Jason2866, 2021-11-23:
@KevinESP32  yes, there has to be a linking error since your platformio setup is wrong.
Do not add any toolchain. The setup i provided does work. Adding toolchains will break.

Jason2866, 2021-12-28:
This build uses Arduino Core 2.0.2 and IDF 4.4
Let's hope there is soon a official release from PlatformIO crew.
Arduino ESP v2.0.2 is a stable release!
```
[env:esp32x]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
board = ...
...

outathebox, 2022-01-03:
@Jason2866 - fantastic!  this worked for me with serial upload,  But trying to use JTAG and tool manager failed to install openocd:

            Tool Manager: Installing git+https://github.com/espressif/openocd-esp32.git
            .
            .
            .
            Error: Could not find one of 'package.json' manifest files in the package

not smart enuf to fix this so i worked around this by adding a platform package to the platform.ini file:
platform_packages = tool-openocd-esp32

Jason2866, 2022-01-03:
@outathebox  thx. Fixed.

outathebox, 2022-01-04:
@Jason2866  so now JTAG upload works for a release build , but for a debug build , there is no debug upload and debugger never starts - no error messages. It was working with:

      platform = espressif32
      platform_packages = framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0 

i'm using the Espidf-Arduino Blink project for testing

Jason2866, 2022-01-04:
Sorry, I can not invest more time in this. Doing a build where all features are working is to  much to do. Probably the latest openocd-esp32 is needed. Providing actual OCD needs packaging the bundle for all OS and uploading to the Platformio registry. This is redundant work, since Platformio or Espressif will do when official release is done.
What I have made so far, we needed for our project so I decided to make it available for all.

outathebox, 2022-01-04:
@Jason2866 Yes i totally understand, I did not expect you to fix it. Thanks very much for making this available  ...forces me to write bug free code :)

viniciusdesa, 2022-01-05:
Looking forward for the arduino-esp32 version 2.0.0! Is there an ETA?

tablatronix, 2022-01-10:
I just realized this @Jason2866 fork, Hope this gets done soon

Plaenkler, 2022-01-14:
Is there a timeline when Arduino v2.0 will be supported? Or a statement from the development team?

mstegen, 2022-01-14:
The build version from @jason2866 works, but in order to use I/O pins 16 and 17, i had to set them up the IDF way.
(pinMode(16, INPUT) does not work)

Jason2866, 2022-01-14:
@mstegen  Correct since it is a fork for Tasmota we have some modifications in the sdkconfig file. We have enabled PSRAM (GPIO 16 17), if none found the Gpios 16 and 17 needs to be reinit to be used.

nickyonge, 2022-01-20:
Commenting to follow. Very excited for ESP32-C3 Arduino!... whenever it comes :')

Simonefardella, 2022-01-21:
@valeros, you have some news about the new official release???

Plaenkler, 2022-01-21:
I tried the workaround from @Json2866 but I get the following error:

```
c:/users/plaenkler/.platformio/packages/toolchain-riscv-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: error: C:\Users\Plaenkler\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32c3\lib\libpthread.a(pthread.c.obj): conflicting priv spec version (major/minor/revision).
```

Do you have a suggestion for a solution? I would really like to port my epaper project.

```
[env:esp32dev]
board = esp32-c3-devkitm-1
framework = arduino
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
	platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
lib_deps = 
	zinggjm/GxEPD2@^1.4.4
	adafruit/Adafruit BusIO@^1.10.1
	adafruit/RTClib@^2.0.2
	bblanchon/ArduinoJson@^6.19.1
        adafruit/Adafruit GFX Library@^1.10.7
	arduino-libraries/NTPClient@^3.1.0
        Wire
	SPI
```

Jason2866, 2022-01-21:
Try latest setup. Before delete all in hidden folder `.platformio`
```
[env:esp32dev]
board = esp32-c3-devkitm-1
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.1/platform-tasmota-espressif32-2.0.2.1.zip
lib_deps = 
	zinggjm/GxEPD2@^1.4.4
	adafruit/Adafruit BusIO@^1.10.1
	adafruit/RTClib@^2.0.2
	bblanchon/ArduinoJson@^6.19.1
        adafruit/Adafruit GFX Library@^1.10.7
	arduino-libraries/NTPClient@^3.1.0
        Wire
	SPI

Plaenkler, 2022-01-21:
> Try latest setup. Before delete all in hidden folder .platformio ....

Worked thank you!

sblantipodi, 2022-01-28:
@ivankravets is there some comments on this issue? is there someone working on it or PlatformIO stopped working on one of the most popular board?

ivankravets, 2022-01-28:
Hi there,

Sorry for the delay. We work with Espressif Team on solving these issues. We will be back with updates after the Chinese New Year holidays.

cc: @pedrominatel

Jason2866, 2022-01-28:
Beeing curios, which issues? Arduino Core 2.0.2 does work nice with platformio.
Or is it just the process of the release? Maybe now done from espressif (as for the toolchains)?

viniciusdesa, 2022-01-28:
> Hi there,
> 
> Sorry for the delay. We work with Espressif Team on solving these issues. We will be back with updates after the Chinese New Year holidays.
> 
> cc: @pedrominatel

Thank you Ivan! If I can contribute in any way, please let me know.
Regards!

fvanroie, 2022-01-28:
I have to manually copy/rename `xtensa-esp32s2-elf-objcopy.exe` to `xtensa-esp32-elf-objcopy.exe` *(without s2)* on an ESP32-S2 build using:
```ini
[arduino_esp32s2_v2]
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.1/platform-tasmota-espressif32-2.0.2.1.zip

board_build.embed_files =
    data/edit.htm.gz
    data/style.css.gz
    data/script.js.gz
```

otherwise I get an error:

```
PLATFORM: Espressif 32 (2.0.2) > Espressif Generic ESP32-S2
HARDWARE: ESP32S2 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.0.2+1
 - tool-esptoolpy 1.30200.211025 (3.2.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
...
Building in release mode
=====================================
Converting .pio\build\wt32-86_ili9488\edit.htm.gz.txt.o
'xtensa-esp32-elf-objcopy' is not recognized as an internal or external command,
operable program or batch file.
Converting .pio\build\wt32-86_ili9488\style.css.gz.txt.o
'xtensa-esp32-elf-objcopy' is not recognized as an internal or external command,
operable program or batch file.
Converting .pio\build\wt32-86_ili9488\script.js.gz.txt.o
'xtensa-esp32-elf-objcopy' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\wt32-86_ili9488\libbf7\LovyanGFX\lgfx\Fonts\IPA\lgfx_font_japan.c.o
```
It is looking for `xtensa-esp32-elf-objcopy.exe` instead of `xtensa-esp32s2-elf-objcopy.exe` in the `.platformio\packages\toolchain-xtensa-esp32s2\bin\` folder:

[image]

Jason2866, 2022-01-28:
@fvanroie  My fork seems to be used :-)  We use this setup in Tasmota and have until now no feedback not working. Probably your setup is broken. 
A Github Actions run for Tasmota ESP32-S2 https://github.com/Jason2866/Tasmota-build/runs/4983324882?check_suite_focus=true

maxgerhardt, 2022-01-28:
No @fvanroie is correct, there is a bug in 

https://github.com/tasmota/platform-espressif32/blob/13ff15cad4abca0540daf20580bd7f1947e6b40b/builder/frameworks/_embed_files.py#L111

because it always uses `xtensa-esp32-elf-objcopy`. That should be replaced with a dynamic evaluation of the toolchain package like https://github.com/tasmota/platform-espressif32/blob/Arduino2022/IDF44/builder/main.py#L157 does, i.e., 

```py
"xtensa-%s-elf-objcopy" % board.get("build.mcu", "esp32")
```

Jason2866, 2022-01-28:
@maxgerhardt  Thx a lot. I copied the orig. Platformio bug.
Do you provide a PR to the Platformio repo?

[46 more replies]

commit 581c7d03 (2022-04-14): Add support for Arduino v2.0

Resolve #619, resolve #652, resolve #591
closed 2022-04-14 by valeros