platformio/platform-espressif32 #620: Wrong arduino-es32 branch

ToMe25, 2021-09-09:
The branches `feature/arduino-idf-master`, `feature/arduino-idf-v4.2`, and `feature/arduino-upstream` all use the [arduino-esp32](https://github.com/espressif/arduino-esp32) branch `master`.

For `feature/arduino-upstream` this is fine.
However for `feature/arduino-idf-master` I believe it should be the [arduino-esp32](https://github.com/espressif/arduino-esp32) branch `idf-master` again.

As for `feature/arduino-idf-v4.2` I am not sure what the correct thing to do would be.
[arduino-esp32](https://github.com/espressif/arduino-esp32) `master` uses v4.4 now afaik.
There is however now a [arduino-esp32](https://github.com/espressif/arduino-esp32) `idf-release/v3.3` branch, so maybe change the `feature/arduino-idf-v4.2` to a v3.3 branch?

Jason2866, 2021-09-09:
The branch `idf-master` is often deleted and imho only a test branch. Merged (and deleted) in the master branch if it works as expected.

ToMe25, 2021-09-09:
> The branch `idf-master` is often deleted and imho only a test branch. Merged (and deleted) in the master branch if it works as expected.

Well, I understand it a bit differently, but more or less its like that.
But isn't testing also what the `feature/arduino-idf-master` branch is for?
If it isn't using that branch I honestly see no reason for its continued existence.

> Pointing idf4.x to a idf 3.x branch is not a good idea

Obviously.
I'm sorry I forgot to mention, but I assumed that branch would be renamed anyways since there probably isn't a good way to point it to v4.2 anymore.
And then I thought that maybe instead of just renaming it to v4.4 renaming it to v3.3 and changing the arduino-esp32 branch might makes sense.

closed 2022-09-20 by stale[bot]