platformio/platform-espressif32 #622: Boot loop on ESP32-C3

agners, 2021-09-12:
I am using the `feature/arduino-idf-master` and the `arduino` framework for ESP32-C3. At one point this used to work, but sometime in early August it seems something broke. Today I had a chance to look a bit more closely, and it seems that the current board config for `esp32-c3-devkitm-1` leads to a boot loop (at least on my board):

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd6100,len:0x1428
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd6100,len:0x1428
ets_loader.c 78
```

It seems that using `DIO` via:

```
board_build.flash_mode = dio
```

does resolve the issue. I am pretty sure however that QIO at one point worked....

martgras, 2022-01-12:
Still fails with 5. if using esp-idf because https://github.com/platformio/platform-espressif32/pull/623 is not merged into master

lukedukeus, 2022-02-23:
I am still having this issue, is this working for anyone else? can you post you upload config?

agners, 2022-02-23:
@lukedukeus setting `board_build.flash_mode` explicitly works around the problem:

```yaml
esphome:
  ...
  platformio_options:
    board_build.flash_mode: dio
```

lukedukeus, 2022-02-23:
That seems to work, but now I'm getting a new error

Compile log:
```
INFO Reading configuration /config/esphome/c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing c3 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 has been installed!
Tool Manager: Installing platformio/framework-espidf @ ~3.40302.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf @ 3.40302.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.16.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-cmake @ 3.16.4 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-ninja @ 1.7.1 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Installing ESP-IDF's Python dependencies
Collecting future>=0.15.2
  Downloading future-0.18.2.tar.gz (829 kB)
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Building wheels for collected packages: future
  Building wheel for future (setup.py): started
  Building wheel for future (setup.py): finished with status 'done'
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=59924f667f4c5b58944307a9d9a6e1abbf2aecf3d4992486cb67dfc5cfea09d7
  Stored in directory: /root/.cache/pip/wheels/2f/a0/d3/4030d9f80e6b3be787f19fc911b8e7aa462986a40ab1e4bb94
Successfully built future
Installing collected packages: pyparsing, future
Successfully installed future-0.18.2 pyparsing-2.3.1
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/api_connection.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/api_frame_helper.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/api_pb2.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/api_pb2_service.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/api_server.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/list_entities.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/proto.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/subscribe_state.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/user_services.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/api/util.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/esp32/core.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/esp32/gpio_arduino.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/esp32/gpio_idf.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/esp32/preferences.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/logger/logger.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/md5/md5.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/mdns/mdns_component.o
Compiling /data/c3/.pioenvs/c3/src/esphome/components/mdns/mdns_esp32_arduino.o
[... 913 lines cut ...]
mode:DIO, clock div:1
load:0x3c090020,len:0x1b780
load:0x3fc8de00,len:0x31f8
load:0x40380000,len:0x1670
load:0x42000020,len:0x80fb4
load:0x40381670,len:0xc604
load:0x50000000,len:0x10
SHA-256 comparison failed:
Calculated: 1c6c73d61b94cb3a21d2d418f11f3c8526e39b0cf81a5acae05e83fcc3351891
Expected: b83ed1f9fb73c74ade2dde861263639a4f6819a5c85edcd3432a74985d756ea1
Attempting to boot anyway...
entry 0x403809b0
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c090020,len:0x1b780
load:0x3fc8de00,len:0x31f8
load:0x40380000,len:0x1670
load:0x42000020,len:0x80fb4
load:0x40381670,len:0xc604
```

Any clue why this is being caused / how to fix it?

agners, 2022-02-23:
I remember I have seen this, but I don't know how I solved it :sweat_smile: 

Tried the same config, and it seems to work here on a ESP32-C3 DevKitM:

```
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x19b0
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2dc0
entry 0x403ce000
I (30) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (30) boot: compile time 16:16:51
I (30) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (44) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 app0             OTA app          00 10 00010000 001c0000
I (99) boot:  4 app1             OTA app          00 11 001d0000 001c0000
I (107) boot: End of partition table
I (111) boot_comm: chip revision: 3, min. application chip revision: 0
```

That is with current dev, flashed using `esphome upload config/c3.yaml`.

How do you flash?

Maybe `version: latest` helps?

lukedukeus, 2022-02-23:
This is still happening even if I am just uploading a basic sketch that prints out to console. 
Is the error that is causing the reboot that the sha checksums arent matching? I am having trouble what is actually causing it to bootloop

[image]

from the simple sketch, it uploads and runs, but still the same error, but with esphome it uploads, but doesnt run.
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000

test
test
test
```

467815891a, 2022-03-01:
I confirm the same problem
here's my esphome config
`
esphome:
  name: esp32c3
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    version: latest
    type: esp-idf
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
    advanced:
      ignore_efuse_mac_crc: false

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "977c2cfaac05fe1bdf6a5e10e3cdec0d"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32C3 Fallback Hotspot"
    password: "eOFyugA8aZOH"
`

it uploads, but doesnt run.

`
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c090020,len:0x1b948
load:0x3fc8de00,len:0x3210
load:0x40380000,len:0x1490
load:0x42000020,len:0x80fb8
load:0x40381490,len:0xc7e4
load:0x50000000,len:0x10
SHA-256 comparison failed:
Calculated: ea122b741a389795f9a41526db2fab1ada89cf3d1b60c6433b5b4ba0d78bf412
Expected: 95d2ecb3766304ca0ae608cfe9d45faf1ded048213bce8835ef050e19913f431
Attempting to boot anyway...
entry 0x403809b0
`

lukedukeus, 2022-03-01:
What board are you using?

valeros, 2022-05-04:
Should be fixed in the latest version of the dev-platform. Please reopen if the problem persists.

closed 2022-05-04 by valeros

platima, 2022-05-04:
Setting 
```
  platformio_options:
    board_build.flash_mode: dio
```
Worked for me

sunshine-nick, 2022-06-16:
I had this boot loop problem with the TTYGO C3 board and the simplest yaml script
I had to do a complete clean set DIO mode force the variant C3 and make sure I was setting latest IDF version
after that it worked. Using "reccomended" still does not work.

esphome:
  name: c3test1
  platformio_options:
    board_build.flash_mode: dio

# Example configuration entry
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
    version: latest
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: false

platima, 2022-06-18:
Hey interesting. I do not have the line "variant: ESP32C3". Will add that to all of them in case it helps!

Freyert, 2022-08-07:
Ran into this issue myself. I was trying to do this all through the home assistant plugin, but it's so slow that I went for using the CLI and had a much better experience.

```
brew install esphome
brew install platformio
platformio upgrade
```

And then I added the:

```yaml
esphome:
  platformIo_options:
    board_build.flash_mode: dio
```

Worked fine after that, wish I understood what happend.

platima, 2022-08-07:
> From "Fulton Byrne" ***@***.***>
> To "platformio/platform-espressif32"
> Cc "platima" ***@***.***>; "Comment"
> Date 8/08/2022 12:58:51 AM
> Subject Re: [platformio/platform-espressif32] Boot loop on ESP32-C3
> (#622)

Weird! Must be a platformio issue for sure.

DamianSuess, 2022-10-24:
Issue just popped up for me over the weekend. The odd thing is, my `sdkconfig.esp32-c3-devkitm-1` file does have the following configuration settings for `DIO`, but I'm getting the exact same error stated by this issue.

```
# CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASH_SAMPLE_MODE_STR=y
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
...
# CONFIG_FLASHMODE_QIO is not set
# CONFIG_FLASHMODE_QOUT is not set
CONFIG_FLASHMODE_DIO=y
```

andychess, 2022-11-11:
I'm getting the same thing with a Seeed Xiao ESP32C3.  

board_build.flash_mode = dio stopped the loop, but I still don't get any output from a hello world example.

My setup is: Mac OS 13.0, VS Code, PlatformIO 6.1.6a2.

Jason2866, 2022-11-11:
@andychess Seems the board uses the inbuilt USB modem (CDC).
Use a C3 boards json which is a CDC device too, or set the mode to use CDC for serial output manually.

HenniePeters, 2023-08-06:
I had a boot loop problem with the development board connected to the PC USB connector. That was a power problem. I could solve it by soldering a 1500uF capacitor over the 3.3V pins.