platformio/platform-espressif32 #627: Partition table parsing wrong

maxgerhardt, 2021-09-15:
Per [community topic](https://community.platformio.org/t/esp-idf-unable-to-mount-see-files-in-platformio-spiffs-image/23454) and https://github.com/ZacDaMan/ESP-IDF-SPIFFS-test.

Consider the following: A PlatformIO ESP-IDF project is given the partition table  

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        0x200000,
storage,  data, spiffs,  ,        1M,
```

We build the uplication and upload it to the microcontroller. The ESP-IDF bootup messages prints how ESP-IDF interprets the partition table.

```
(51) boot: Partition Table:␛[0m
(55) boot: ## Label            Usage          Type ST Offset   Length
(62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(77) boot:  2 factory          factory app      00 00 00010000 00200000
(85) boot:  3 storage          Unknown data     01 82 00210000 00100000
(92) boot: End of partition table␛[0m
```

However when we now upload SPIFFS data via PlatformIO (`pio run -t uploadfs`) we'll see 

```
Wrote 1048576 bytes (52274 compressed) at 0x00207000 in 7.2 seconds (effective 1160.7 kbit/s)...
```

So PlatformIO has interpreted the address of the SPIFFS partition wrong, as `0x00207000` vs ESP-IDF's `0x00210000` (difference 0x3000).

This is because in this case of the partition table having no offset information, PlatformIO uses an algorithm for determining the offset / address by summing up the sizes of all previous partitions

https://github.com/platformio/platform-espressif32/blob/1e0f14730b041a337efd9f37d915851d703ee041/builder/main.py#L57-L76

There is however a special case that this algorithm does not respect: Per [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#)

>Partitions of type app have to be placed at offsets aligned to 0x10000 (64K). If you leave the offset field blank, gen_esp32part.py will automatically align the partition. If you specify an unaligned offset for an app partition, the tool will return an error

In this case PlatformIO has determined the address of the factory app by summing the sizes of the previous two partitions (0x6000 + 0x1000 = 0x7000), but the real address of the factory partition is, due to the 0x10000 alignment above, 0x10000 (up-rounded to the nearest multiple of 0x10000).

Thus a 0x3000 difference is introduced that messes up the final SPIFFS partition address, PlatformIO uploads to the wrong address and the SPIFFS mounting code doesn't find a SPIFFS partition data at the address it expects (`0x00210000`).

The issue can be corrected for PlatformIO by adding an explicit offset information for the factory partition.

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, 0x10000, 0x200000,
storage,  data, spiffs,  ,        1M,
```

This causes the interpretation in the ESP-IDF firmware tostill be **exactly** the same with the same output above, but now PlatformIO does the correct 

```
Wrote 1048576 bytes (52274 compressed) at 0x00210000 in 7.0 seconds (effective 1205.3 kbit/s)...
```

for a filesystem upload

TL;DR: PlatformIO must internally correct the offset for all `app` type partitions to be 0x10000 aligned, or the partition table interpretation of ESP-IDF and PlatformIO will be different.

maxgerhardt, 2021-09-15:
Addendum: I've also noticed that the initial offset value in PlatformIO is 0 but the first partition in ESP-IDF starts at `0x9000`. Should be investigated.

ZacDaMan, 2021-09-15:
The initial offset may simply be how the alignment is implemented, adding the unused bytes before the first partition instead of between `phy_init` and `factory`. Would be interesting to see what happens if a second `app` type partition is added when the sizes don't align.

ZacDaMan, 2021-09-16:
I tried adding a second misaligned `app` partition and got these results.

My partitions.csv file:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,        data, nvs,                    ,        0x6000,
phy_init, data, phy,                    ,        0x1000,
factory,   app,  factory,               ,        0x204000,
ota_0,     app,  ota_0,                  ,        0x13000,
storage,  data, spiffs,  0x250000,        1M,
```

What ESP-IDF compiled:
```
# Name,     Type,   SubType,    Offset,     Size, Flags
nvs,            data,   nvs,           0x9000,     24K (~0x6000),
phy_init,     data,   phy,          0xf000,      4K (~0x1000),
factory,       app,    factory,    0x10000,    2064K (~0x204000),
ota_0,         app,    ota_0,       0x220000,   76K (~0x13000),
storage,     data,   spiffs,        0x250000,   1M,
```

So the first 'gap' to align `factory` appears to have been added before the first partition, while the second gap to align `ota_0` was added after `factory`. 

Swapping `storage` and `ota_0`, we get:
```
# Name, Type, SubType, Offset, Size, Flags
nvs,        data,   nvs,       0x9000,      24K,
phy_init, data,   phy,       0xf000,      4K,
factory,   app,    factory, 0x10000,    2064K,
storage,  data,   spiffs,    0x214000,  1M,
ota_0,      app,   ota_0,    0x320000,   76K,
```
Here the second gap is again inserted directly before `ota_0` rather than before `storage` - potentially only the very first offset is added at the start, with additional offsets being inserted directly before their respective partitions?

commit c4ad0942 (2021-11-05): Properly calculate partition offsets

Resolve #627
closed 2021-11-05 by valeros