platformio/platform-espressif32 #629: Add debug support for M5Stack Core2

Zokol, 2021-09-18:
### Description of problem
Uploading code works fine, but I would need the debug-functionality.
This would greatly increase the usability of this platform, as it is quite packed with peripherals, so the codebase can easily get large and complex.

### Additional info
There is no direct interface for JTAG in Core2.
You will need to solder two wires, for TDI and TDO (GPIO 12 and GPIO 15). These pins are used for LCD and speaker IO.
TMS and TCK are already wired to the M-BUS-connector, and TDI/TDO pins can be wired to unused pins of the connector.

Few pics of the required mods:

[image]
[image]

Jason2866, 2021-09-18:
Easy to solve. Just do your own boards.json defintion for your modified core2 device and place it in folder `boards` in the root of your platformio project.

li3p, 2022-05-08:
Any updates on this thread?

closed 2022-08-11 by stale[bot]