platformio/platform-espressif32 #630: ESP32 no longer works with Flash Mode QIO, continual rebooting, must use QOUT or DIO

RalphBacon, 2021-09-27:
Until recently I always used Flash Mode QIO at 80MHz on all my ESP32 boards - all fine.

Recently, possibly after upgrade to PlatformIO Arduino Framework 3.3.2, the QIO mode always gives a continuous reboot, even on boards that have previously been successfully running for a long time with QIO mode.

**Continuous reboot trace:**
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:04:41.152 --> configsip: 0, SPIWP:0xee
15:04:41.152 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:04:41.152 --> **mode:QIO**, clock div:1
15:04:41.152 --> load:0x3fff0018,len:4
15:04:41.152 --> load:0xfe2520a2,len:547496191
15:04:41.152 --> 1162 mmu set 00010000, pos 00010000
15:04:41.178 --> 1162 mmu set 00020000, pos 00020000
15:04:41.191 --> 1162 mmu set 00030000, pos 00030000
15:04:41.214 --> 1162 mmu set 00040000, pos 00040000
15:04:41.231 --> 1162 mmu set 00050000, pos 00050000
15:04:41.253 --> 1162 mmu set 00060000, pos 00060000
15:04:41.272 --> 1162 mmu set 00070000, pos 00070000
15:04:41.294 --> 1162 mmu set 00080000, pos 00080000
15:04:41.318 --> 1162 mmu set 00090000, pos 00090000
15:04:41.341 --> 1162 mmu set 000a0000, pos 000a0000
15:04:41.355 --> 1162 mmu set 000b0000, pos 000b0000
15:04:41.372 --> 1162 mmu set 000c0000, pos 000c0000
15:04:41.392 --> 1162 mmu set 000d0000, pos 000d0000
15:04:41.415 --> 1162 mmu set 000e0000, pos 000e0000
15:04:41.446 --> 1162 mmu set 000f0000, pos 000f0000
15:04:41.460 --> 1162 mmu set 00100000, pos 00100000
15:04:41.483 --> 1162 mmu set 00110000, pos 00110000
15:04:41.498 --> ets Jun  8 2016 00:22:57

**Using QOUT gives the (expected) result:**
15:04:03.821 --> ets Jun  8 2016 00:22:57
15:04:03.821 --> 
15:04:03.821 --> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:04:03.821 --> configsip: 0, SPIWP:0xee
15:04:03.821 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:04:03.821 --> **mode:QOUT**, clock div:1
15:04:03.821 --> load:0x3fff0018,len:4
15:04:03.821 --> load:0x3fff001c,len:1216
15:04:03.833 --> 
15:04:03.833 --> ho 0 tail 12 room 4
15:04:03.833 --> load:0x40078000,len:10944
15:04:03.833 --> load:0x40080400,len:6388
15:04:03.833 --> entry 0x400806b4
15:04:03.961 --> **Sucessful start with flash mode: qout**

Using test board TTGO T7 v1.3 on PlatformIO using Arduino framework 3.3.2 under Windows 10 (but it seems to make no difference which board I try this on, they ALL exhibit this behaviour now).

**PlatformIO.ini settings:**
[upload_settings]
board_build.flash_mode = qio

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_build.f_flash = 80000000L
upload_port=COM4
upload_speed=921600
build_flags =
    -D FLASHMODE=\"${upload_settings.board_build.flash_mode}\"
extends = upload_settings

**Sketch:**
`#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
    pinMode(16, OUTPUT);
    digitalWrite(16, LOW);
    printf("Sucessful start with flash mode: %s\n", FLASHMODE);
}

void loop()
{
    digitalWrite(16, millis() % 1000 > 500 ? HIGH : LOW);
}`

Has there been in change in the Espressif code to account for this anomaly?

RalphBacon, 2021-09-27:
I've read this article many times but it does not shed any light on this unexpected behaviour (was working / now isn't).
https://github.com/espressif/esptool/wiki/SPI-Flash-Modes

cosmoiler, 2021-09-28:
A similar problem with versions 3.3.1 and 3.3.2. With version 3.3.0, everything works fine.

RalphBacon, 2021-09-28:
I regressed to version 3.3.0 to test this and you are correct. 

Under 3.3.0 even though I specify QIO in my PlatformIO.ini, on startup I get this mode:

14:33:43.187 --> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:33:43.187 --> configsip: 0, SPIWP:0xee
14:33:43.187 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:33:43.187 --> **mode:DIO**, clock div:1
14:33:43.187 --> load:0x3fff0018,len:4
14:33:43.187 --> load:0x3fff001c,len:1216
14:33:43.187 --> ho 0 tail 12 room 4
14:33:43.187 --> load:0x40078000,len:10944
14:33:43.187 --> load:0x40080400,len:6388
14:33:43.187 --> entry 0x400806b4
14:33:47.066 --> Checking SSID 

Board parameters:
board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
board_build.flash_mode = qio

So it appears they might have fixed a bug where QIO was being treated as DIO but now causes failures because QIO cannot be handled by any of the ESP32 modules I have (I've tested about 10, all different, although some from the same manufacturer TTGO).

Whatever, it's a bug and should be acknowledged by Espressif with an explanation (even if the workaround is to use QOUT or buy better boards with flash memory that can handle QIO).

This has caused me endless hours of needless problem solving.

closed 2021-11-05 by valeros

raphael-bmec-co, 2024-08-20:
It seems as though this may not be totally resolved (or we are doing something wrong).

### Steps to reproduce:

`dio`, and `dout` behave as expected. We only see the issue on `qio`. 

`platform = platformio/espressif32@6.7.0`

1. Set the flash `board_build.flash_mode = qio` in `platformio.ini`.
2. Build
3. Run `esptool.exe --chip esp32 image_info {bootloader/firmware}.bin --version 2`

### Expected output:
`Flash mode: QIO`

### Actual output:
`Flash mode: DIO`

bootloader.bin full output:
```
esptool.py v4.7.0
File size: 18992 (bytes)

ESP32 image header
==================
Image version: 1
Entry point: 0x400805f0
Segments: 3
Flash size: 4MB
Flash freq: 80m
Flash mode: DIO

ESP32 extended image header
===========================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 0 (ESP32)
Minimal chip revision: v0.0, (legacy min_rev = 0)
Maximal chip revision: v655.35

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x00540  0x3fff0030  0x00000018  BYTE_ACCESSIBLE, DRAM, DIRAM_DRAM
      2  0x0368c  0x40078000  0x00000560  CACHE_APP
      3  0x00e10  0x40080400  0x00003bf4  IRAM

ESP32 image footer
==================
Checksum: 0xf6 (valid)
Validation hash: 43727af40a76c6f683c141a3dea94fe9fd9c3924ec81aa2ec16e768730ad4193 (valid)
```
firmware.bin full output:
```
esptool.py v4.7.0
File size: 1750992 (bytes)

ESP32 image header
==================
Image version: 1
Entry point: 0x40083464
Segments: 5
Flash size: 4MB
Flash freq: 80m
Flash mode: DIO

ESP32 extended image header
===========================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 0 (ESP32)
Minimal chip revision: v0.0, (legacy min_rev = 0)
Maximal chip revision: v655.35

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x37f00  0x3f400020  0x00000018  DROM
      2  0x06dc4  0x3ffbdb60  0x00037f20  BYTE_ACCESSIBLE, DRAM
      3  0x01324  0x40080000  0x0003ecec  IRAM
      4  0x14fbd0  0x400d0020  0x00040018  IROM
      5  0x1bbac  0x40081324  0x0018fbf0  IRAM

ESP32 image footer
==================
Checksum: 0x5d (valid)
Validation hash: cba496eba3d4ee9f654f9996bcd8bc7a9c219b090d3e509e3db5e4fcce48a1f4 (valid)

Application information
=======================
Project name: arduino-lib-builder
App version: esp-idf: v4.4.7 38eeba213a
Compile time: Mar  5 2024 12:36:24
ELF file SHA256: 3caeb51962f51284a90cbd4665e4e51da34d5b5899143e9880b57cb21193c688
ESP-IDF: v4.4.7-dirty
Secure version: 0
```

raphael-bmec-co, 2024-08-20:
@Jason2866 forgive my ignorance on this topic. If there is some documentation that answers the questions below please just point us there.

[This](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/startup.html) seems to indicate that the `bootloader.bin` is the 2nd state bootloader?

> In ESP-IDF, the binary image which resides at offset 0x1000 in flash is the second stage bootloader.

1. Is `board_build.flash_mode = qio/qout` in platformio.ini ignored or included in the second stage bootloader?
2. Are there other steps that need to be taken to set flash mode qout/qio?
3. We are trying to validate that the settings in platform.ini are making it through to the build files. Any idea if/where we can validate that qout/qio is being set, or can this only be done at runtime?

Jason2866, 2024-08-20:
1. it is taken in account during the build process
2. No
3. If it is really running in mode QIO can only be validated during runtime

Documentation https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/spi-flash-modes.html#frequently-asked-questions

commit f7bcd7e6 (2021-11-05): Revert "Use default flash mode when uploading via esptoolpy"

Resolve #584, resolve #630, resolve #623

This reverts commit fe5d2796d46cc4fb29f8a77cbfa66f53faf4fb89.