platformio/platform-espressif32 #632: Please update IDF to stable v4.3.1

Jason2866, 2021-09-28:
It is released since a while https://github.com/espressif/esp-idf/releases/tag/v4.3.1

liebman, 2021-10-16:
> ```
> platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
> platform_packages  = framework-espidf @ https://github.com/espressif/esp-idf.git#v4.3.1
> ```
For me generates errors:
```
➜ pio run
Processing liebman (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master; board: esp32dev; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 130, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 134, in cli
    process_env(
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/commands/platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "/Users/chris.l/.platformio/penv/lib/python3.8/site-packages/platformio/platform/factory.py", line 70, in new
    platform_cls = getattr(
AttributeError: module 'platformio.platform.platform' has no attribute 'PlatformPlatform'
```

valeros, 2021-11-17:
Updated in [v3.4.0](https://github.com/platformio/platform-espressif32/releases/tag/v3.4.0)

closed 2021-11-17 by valeros