platformio/platform-espressif32 #635: Compilation error due to missing x509_crt_bundle

gohm44, 2021-09-30:
Step to reproduce:

Create new project and compile it.
1. mkdir /tmp/platformio_project
2. virtualenv venv
3. venv/bin/pip install platformio
4. venv/bin/pio project init --board esp-wrover-kit --project-option "framework=espidf" --ide vim
5. venv/bin/pio run

Observed behaviour:

1. At the beginning there is error which does not brake compilation yet:
```
Reading CMake configuration...
Generating assembly for certificate bundle...
CMake Error at /home/gohm/.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
  file failed to open for reading (No such file or directory):

    /tmp/platformio_project/.pio/build/esp-wrover-kit/x509_crt_bundle

*** Error 1
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
```

But finally compilation failed due to missing `.pio/build/esp-wrover-kit/x509_crt_bundle.S`
```
Compiling .pio/build/esp-wrover-kit/lwip/port/esp32/netif/ethernetif.o
Compiling .pio/build/esp-wrover-kit/lwip/port/esp32/vfs_lwip.o
Compiling .pio/build/esp-wrover-kit/mbedtls/esp_crt_bundle/esp_crt_bundle.o
*** [.pio/build/esp-wrover-kit/.pio/build/esp-wrover-kit/x509_crt_bundle.o] Source `.pio/build/esp-wrover-kit/x509_crt_bundle.S' not found, needed by target `.pio/build/esp-wrover-kit/.pio/build/esp-wrover-kit/x509_crt_bundle.o'.
====================== [FAILED] Took 20.48 seconds ======================
```

bartezz32, 2021-09-30:
Temporary fix for this problem [issue-1012039784](https://github.com/espressif/esp-idf/issues/7621#issue-1012039784)

manuelbl, 2021-09-30:
Could it be the *DST Root CA X3* certificate?

```
        Issuer: O=Digital Signature Trust Co., CN=DST Root CA X3
        Validity
            Not Before: Sep 30 21:12:19 2000 GMT
            Not After : Sep 30 14:01:15 2021 GMT
        Subject: O=Digital Signature Trust Co., CN=DST Root CA X3
```

It just expired today....

commit 93b71b66 (2021-11-05): Update ESP-IDF to v4.3.1

Resolve #655, resolve #635
closed 2021-11-05 by valeros