platformio/platform-espressif32 #637: pyparsing: Unable to parse section

malja, 2021-10-01:
Hi guys, I have a ESP project with following settings:

**platformio.ini**

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

debug_tool = esp-prog
debug_init_break = tbreak app_main
upload_protocol = esp-prog
monitor_speed = 115200

build_flags =
    -DCOMPONENT_EMBED_FILES=components/ss_mqtt/certs/ca.pem:components/ss_mqtt/certs/client.pem:components/ss_mqtt/certs/key.pem
    -Og
    -ggdb3
build_unflags = -Os
```

**pio system info**

```
--------------------------  --------------------------------------------
PlatformIO Core             5.2.1b2
Python                      3.9.2-final.0
System Type                 darwin_x86_64
Platform                    macOS-10.16
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/username/.platformio
PlatformIO Core Executable  /Users/username/.platformio/penv/bin/platformio
Python Executable           /Users/username/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          12
--------------------------  --------------------------------------------
```

**pio platform update**

```
Platform espressif32
--------
Updating platformio/espressif32               3.3.2                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32ulp        1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating platformio/framework-espidf          3.40300.0 @ ~3.40300.0             [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-openocd-esp32        2.1000.20210721 @ ~2.1000.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-ninja                1.9.0 @ ^1.7.0                     [Up-to-date]
```

**python version**

```
% /Users/username/.platformio/penv/bin/python --version
Python 3.9.2
```

**pyparsing version**

```
% /Users/username/.platformio/penv/bin/python -m pip show pyparsing
Name: pyparsing
Version: 2.3.1
Summary: Python parsing module
Home-page: https://github.com/pyparsing/pyparsing/
Author: Paul McGuire
Author-email: ptmcg@users.sourceforge.net
License: MIT License
Location: /Users/username/.platformio/penv/lib/python3.9/site-packages
Requires: 
Required-by: 
```

**Build process ends with following error message:**

```
Traceback (most recent call last):
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 156, in _get_infos_from_file
    results = parser.parseString(info.content, parseAll=True)
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 1814, in parseString
    raise exc
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 1808, in parseString
    se._parse( instring, loc )
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 3722, in parseImpl
    loc, exprtokens = e._parse( instring, loc, doActions )
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 1552, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/username/.platformio/penv/lib/python3.9/site-packages/pyparsing.py", line 3502, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected end of text (at char 153608), (line:2513, col:50)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 178, in <module>
    main()
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 154, in main
    mapping_rules = generation_model.generate_rules(sections_infos)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 382, in generate_rules
    root_node.insert(entity, sections, target, entities)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 257, in insert
    self.add_node_child(entity, sections, target, sections_db)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 130, in add_node_child
    child.insert(entity, sections, target, sections_db)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 149, in insert
    self.add_node_child(entity, sections, target, sections_db)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 130, in add_node_child
    child.insert(entity, sections, target, sections_db)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 149, in insert
    self.add_node_child(entity, sections, target, sections_db)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/generation.py", line 189, in add_node_child
    obj_sections = sections_db.get_sections(self.parent.name, self.name)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 199, in get_sections
    obj = self._match_obj(archive, obj)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 187, in _match_obj
    objs = self.get_objects(archive)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 180, in get_objects
    self._process_archive(archive)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 168, in _process_archive
    parsed = self._get_infos_from_file(stored)
  File "/Users/username/.platformio/packages/framework-espidf/tools/ldgen/entity.py", line 158, in _get_infos_from_file
    raise ParseException('Unable to parse section info file ' + info.filename + '. ' + p.msg)
pyparsing.ParseException: Unable to parse section info file /Users/username/Projects/SmartSense/main_esp_unit/.pio/build/esp32dev/esp-idf/esp_event/libesp_event.a. Expected end of text (at char 0), (line:1, col:1)
*** [.pio/build/esp32dev/esp32.project.ld] Error 1
esptool.py v3.1
Merged 1 ELF section
============================================================================== [FAILED] Took 103.98 seconds ==============================================================================
```

I tried installing pyparsing [2.2.0 as suggested](https://community.platformio.org/t/esp-idf-pyparsing-expecting-end-of-text/23562), but it did not chage anything. I tried adding [UTF8 support](pyparsing.ParseException: Unable to parse section info file), but as far as I know, it was already added to pyparsing and my path does not contain non-ASCII characters.

Next I tried to update both platformio and esp-idf to latest version. Again, nothing helped.

Do you have any idea what could be done?

HarryVienna, 2022-06-28:
Hello,

I had a similar problem today. Same error as you. In my platform.ini file a had the line

`debug_build_flags = -O0 -ggdb3 -g3`

When I removed that line, it worked again. Unfortunately I have no idea what the reason can be... It worked some months ago, so I perhaps an update broke something?

Harry

closed 2022-08-11 by stale[bot]

jef-sure, 2023-06-21:
I have the same problem:

Successfully created esp32 image.
Traceback (most recent call last):
  File "C:\Users\kymj\.platformio\packages\framework-espidf\tools\ldgen\ldgen\entity.py", line 152, in _get_infos_from_file
    results = parser.parseString(info.content, parseAll=True)
  File "C:\Users\kymj\.platformio\penv\.espidf-5.0.2\lib\site-packages\pyparsing\util.py", line 256, in _inner
    return fn(self, *args, **kwargs)
  File "C:\Users\kymj\.platformio\penv\.espidf-5.0.2\lib\site-packages\pyparsing\core.py", line 1190, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected end of text, found 'default'  (at char 2), (line:2, col:1)

valeros, 2023-06-21:
@jef-sure Should be fixed in the latest version of the dev platform.

jepsenphil, 2024-01-30:
> Hello,
>
> I had a similar problem today. Same error as you. In my platform.ini file a had the line
>
> `debug_build_flags = -O0 -ggdb3 -g3`
>
> When I removed that line, it worked again. Unfortunately I [...]

I'm still getting this issue.   Here is my platformio.ini 

[env]
platform = espressif32@~5.4.0
platform_packages =
	platformio/tool-openocd@^3.1200.0
framework = arduino, espidf
debug_build_flags = -O0 -ggdb3 -g3

File "/home/phil/.platformio/penv/.espidf-4.4.5/lib/python3.10/site-packages/pyparsing.py", line 3502, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected end of text (at char 161023), (line:2559, col:50)