platformio/platform-espressif32 #638: sdkconfig.defaults per environment

MMI, 2021-10-05:
Currently, I can have a `sdkconfig.defaults` for a project.

When I set up a new environment `env:FOO` the build will generate a new `sdkconfig.FOO` derived from `sdkconfig.defaults`.

If I setup a new sub-environment `env:BAR` that contains an `extends = env:FOO`, the build will generate a new `sdkconfig.BAR` derived from `sdkconfig.defaults`.

Awesome. Ish.

What I would like is the ability to supply different `sdkconfig.defaults` to an environment (such that it could be inherited by a sub-environment that uses `extends =`.

vogelfreiheit, 2022-05-04:
Is this supported right now? (using sdkconfig.defaults per the target env, one for each env -different boards-)?

shlomozippel, 2022-12-28:
I was able to hack my way around this but obviously I hope there will be official support for this. My solution for now is to edit the top level `CMakeLists.txt` and add the following before including the espidf project.cmake:

```
cmake_minimum_required(VERSION 3.16.0)

if(EXISTS "${SDKCONFIG}.defaults")
    if(EXISTS "sdkconfig.defaults")
        set(SDKCONFIG_DEFAULTS "${SDKCONFIG}.defaults;sdkconfig.defaults")
    else()
        set(SDKCONFIG_DEFAULTS "${SDKCONFIG}.defaults")
    endif()
    message(STATUS "-- Using defaults: ${SDKCONFIG_DEFAULTS}")
endif()

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(projectname)
```

This lets you use `sdkconfig.<pioenv>.defaults` as well as the standard `sdkconfig.defaults` for common defaults

alexander-rd, 2023-01-02:
I also taggled this during the last days. Came up with a convinient-enough solution based on Pre-Actions script and three custom platform.io options that are even usable with one level of "extends". 

In general, you need to add this script (see attachement).
> [env]
> extra_scripts = pre:config/pioEnvHacksForIdf.py

For any hardware configuration, you add something like:

> [esp32Solo]
> ...
> custom_idfSdkConfigDefaults = config/idf/sdkconfig.defaults
> custom_idfSdkConfig         = config/idf/sdkconfig.esp32Solo

If you have multiple software main files, you also need different CMakeLists.txt for IDF. Here, you use:
> [devTest]
> build_src_filter = +<mainDevTest.cpp>
> custom_idfCMakeLists = config/idf/CMakeLists.DevTest

And a final environment:
> [env:esp32Solo_devTest]
> extends = esp32Solo, devTest

Of course, you don't need to use "extends".
The scripting was tested to fulfill:
> 1. Options from sdkconfig and CMakeLists.txt are applied (for the first
>     build)
>  2. after two environments are built, switching between them does not need to
>     rebuild For the following use cases, the workaround would be "clean".
>  3. Changes in the sdkconfig files are detected and applied
>  4. Changes in the CMakeLists files are detected and applied 
> Results: all good. (3) does only rebuild the main binary while (4) rebuilds everything.

[pioEnvHacksForIdf.zip](https://github.com/platformio/platform-espressif32/files/10332447/pioEnvHacksForIdf.zip)

ldab, 2024-05-23:
It is already possible, by just, for example:

```
[env:release]
build_type = release

board_build.cmake_extra_args = 
    -DSDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.release.defaults"
```

As per https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32/api-guides/build-system.html#custom-sdkconfig-defaults

tavdog, 2025-03-22:
using -DSDKCONIFG_DEFAULTS does work but it doesn't rebuild when building for a new environment. How tel tell pio to force a regeneratino of sdkconfig during build or switching envs ?