platformio/platform-espressif32 #657: EEPROMClass::begin(4096) truncate size fails when previously set to 8Kb

Nagymadar, 2021-10-30:
I have tried to set the EEPROM size to 8K, but failed. I guess it is maximized to 4K?

Nevertheless when I set to 4K again, EEPROM.begin(4096) failed to truncate it back from 8K->4K. 

In EEPROM.cpp:
bool EEPROMClass::begin(size_t size)

It tries to truncate, but these functions fail with ESP_ERR_NVS_NOT_ENOUGH_SPACE: 
nvs_get_blob(_handle, _name, key_data, &key_size);
nvs_set_blob(_handle, _name, key_data, size);
nvs_commit(_handle);

I had to erase the whole chip as workaround. But I think this should have worked.

maxgerhardt, 2021-11-02:
If there are problems with the workings of the Arduino core (double-check the latest 2.0.0 version in the Arduino IDE), please open an issue in https://github.com/espressif/arduino-esp32/issues, not here.

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros