platformio/platform-espressif32 #668: After the update, the menu Platform item in the project tasks was lost.

SinglWolf, 2021-11-08:
In my project I use two boards: `esp-wrover-kit` and `esp32-s2-saola-1`.
After upgrading `PlatformIO` to version `2.4.0`, the `Platform` menu item disappeared in the project tasks for the `esp-wrover-kit` board.
`platfirmio.ini` for `esp32-s2-saola-1`:
```
[platformio]
src_dir = main
include_dir = main/include
description = controlPanel

[env:esp32dev]
platform = espressif32
framework = espidf
; board = esp-wrover-kit
board = esp32-s2-saola-1
monitor_speed = 115200
monitor_filters = direct
```
[image]

It's okay.

`platfirmio.ini` for `esp-wrover-kit`:
```
[platformio]
src_dir = main
include_dir = main/include
description = controlPanel

[env:esp32dev]
platform = espressif32
framework = espidf
board = esp-wrover-kit
; board = esp32-s2-saola-1
monitor_speed = 115200
monitor_filters = direct
```
[image]
`Platform` menu item not.
Removing `PlatformIO` and reinstalling from scratch (with removing the `.platformio` directory) does not help.
What have I done wrong?

ivankravets, 2021-11-08:
May I ask you to remove `C:\\Users\\Alexander\\.platformio\packages` folder and start a build again?

SinglWolf, 2021-11-08:
> May I ask you to remove `C:\\Users\\Alexander\\.platformio\packages` folder and start a build again?

I kind of already talked about this at the very beginning.

> Removing `PlatformIO` and reinstalling from scratch (with removing the `.platformio` directory) does not help.

SinglWolf, 2021-11-08:
When creating a new project with the board `ESP32 Dev Module` `PlatformIO` itself suggested sending [this bug report](https://github.com/platformio/platformio-home/issues/3272).

ivankravets, 2021-11-08:
> PLATFORM: Espressif 32 (3.3.2+sha.824c5fb)

I see, you switched to the upstream version of this dev-platform. It is still in progress and has not been released yet. Could you switch back to the stable version? -> `platform = espressif32`?

SinglWolf, 2021-11-08:
> Could you switch back to the stable version? -> `platform = espressif32`?

This problem occurred `BEFORE` I upgraded to version `--dev`.

https://github.com/platformio/platform-espressif32/issues/668#issuecomment-962451273

SinglWolf, 2021-11-08:
In all my projects I replaced at `platform = espressif32`.
Removed `PlatformIO` completely (with `.platform` directory), removed `VScode` with registry cleanup. Rebooted the computer. Did a re-installation `VScode+PlatformIO`. It works.
What it was, I still did not understand. The issue can be closed.

ivankravets, 2021-11-08:
Frankly speaking, I don't have any ideas :( Please keep us updated if you face this issue again.

Thanks for using PlatformIO! 🙏

closed 2021-11-08 by ivankravets