platformio/platform-espressif32 #670: Add Support for M5STAMP-C3

trrevvorr, 2021-11-18:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

### Description of problem
Please add support for the [M5STAMP-C3](https://docs.m5stack.com/en/core/stamp_c3) SKU:C056-B, K056

### Additional info
There are several other M5STACK boards already supported (e.g. the [M5Stack-ATOM](https://docs.platformio.org/en/latest/boards/espressif32/m5stack-atom.html)). There is also now one ESP32-C3 board supported: [Espressif ESP32-C3-DevKitM-1](https://docs.platformio.org/en/latest/boards/espressif32/esp32-c3-devkitm-1.html)

leCandas, 2021-12-03:
I am also waiting forward for this. Is there any ETA?

leCandas, 2021-12-05:
I have finally made it work see the repo 
https://github.com/leCandas/M5Stamp-C3

juanitomaille, 2022-04-01:
There also M5Stamp-C3U. (C3 discontinued ?)

leCandas, 2022-04-01:
the only difference is c3u does not have a serial chip on, you flash directly to esp32 c3

gmag11, 2022-05-08:
You need CDC serial emulation. Check this: https://github.com/platformio/platform-espressif32/blob/develop/boards/adafruit_feather_esp32s2_tft.json#L10

gmag11, 2022-05-08:
This is the output of `pio device list`

```
Hardware ID: USB VID:PID=303A:1001 SER=60:55:F9:57:B6:10
Description: USB Serial Device (COM6)
```

ThomDietrich, 2022-11-29:
Any confirmed `platformio.ini` configuration for the C3U?
Thanks!

Jason2866, 2022-11-29:
@ThomDietrich  https://github.com/platformio/platform-espressif32/blob/develop/boards/adafruit_qtpy_esp32c3.json
should work since it is the same from board design.

thayoz, 2023-08-02:
Any update ? I have tried to use it today, but this doesn't look to be officially supported.

valeros, 2026-10-17:
Resolved in https://github.com/platformio/platform-espressif32/pull/1491

closed 2026-10-17 by valeros