platformio/platform-espressif32 #671: Cannot find CMAKE_ROOT on Windows

sudoDavi, 2021-11-18:
Hello, I'm pretty new to PlatformIO, I'm just using it to test a project I've found on another repository. So this might be just a misconfiguration on my part, but I can't find how to fix it in the docs or the IDE.

  Anyway, the problem is that the project is failing to build because it can't find `CMAKE_ROOT`, meaning that it can't use cmake to build the project, I've looked into the .platformio folder to see if the cmake package was downloaded and it was.

  I'm on Windows and cmake is located in the folder `packages\tool-cmake\bin` inside the `.platformio` folder, but the build configuration of the project is looking for cmake in the `packages\tool-cmake\share\cmake-3.16` folder, which doesn't exist.

This is the complete output of the build terminal:
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40301.0 (4.3.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r1
 - toolchain-xtensa-esp32 8.4.0+2021r1
 - toolchain-xtensa-esp32s2 8.4.0+2021r1
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
Reading CMake configuration...

CMake Error: Could not find CMAKE_ROOT !!!
CMake has most likely not been installed correctly.
Modules directory not found in
C:/Users/MyUser/.platformio/packages/tool-cmake/share/cmake-3.16
CMake Error: Error executing cmake::LoadCache(). Aborting.
```

valeros, 2021-11-18:
Hi @sudoDavi ! Based on the warning message in your log:
```
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
``` 
The project you're trying to compile seems to be incompatible with the latest ESP-IDF.

sudoDavi, 2021-11-19:
> Hi @sudoDavi ! Based on the warning message in your log:
>
> ```
> Warning! Starting with ESP-IDF v4.0, new project structure is required: [...]

Hi @valeros , thanks for replying.

That's the thing, I've looked at that link and from what I could see the project does attend to the new file structure, as can be seen in this screen shot:
[image]

sudoDavi, 2021-11-23:
> @sudoDavi Any chance you could you please share a link to this project?

Sure, it's the zip file in this issue comment
https://github.com/CANopenNode/CANopenNode/issues/198#issuecomment-658429391

valeros, 2021-11-24:
Strange, but I cannot reproduce the issue on my Windows machine., the project compiled just fine. Could you please delete the folder `C:/Users/MyUser/.platformio/packages/tool-cmake` and try to compile your project again?

sudoDavi, 2021-11-25:
> Strange, but I cannot reproduce the issue on my Windows machine., the project compiled just fine. Could you please delete the folder `C:/Users/MyUser/.platformio/packages/tool-cmake` and try to compile your project again?

Great, this fixed it!!
It's compiling now!!
Thanks @valeros, I know this is a pretty simple fix to an unknown cause, but I think it would be great if it was added somewhere in the docs!

valeros, 2021-11-25:
> I know this is a pretty simple fix to an unknown cause, but I think it would be great if it was added somewhere in the docs!

For some reason your previous installation for the `CMake` packages was corrupted (maybe installation process was interrupted, an antivirus didn't like the package, etc.). Anyway, you are the first one who reported such a problem for the last several years. I believe we can consider this problem as a one-time incident.

closed 2021-11-25 by valeros