platformio/platform-espressif32 #674: Request: ESP32-S3-DevKitC-1 board support

maumad, 2021-11-21:
- [ ] **Feature Request**.
      It would be nice to have support to ESP32-S3-DevKitC-1 - MCU ESP32S3.

Best regards

AndyW999, 2021-12-20:
Is this board request likely to be fulfilled for ESP_IDF?

Thanks

Andy.

darkfiberiru, 2021-12-28:
@AndyW999 are you asking on platformio side or esp_idf side. 4.4 and 5.0 support esp32-s3. 4.4 is in beta and 5.0 is dev branch

AndyW999, 2021-12-29:
ESP-IDF preferably.

jonathandreyer, 2022-01-09:
Hi everyone,

I have also a ESP32-S3-DevKitC-1 and I have started the porting of this chip to PlatformIO (with ESP-IDF v4.4-beta). That was possible by the improvement made by @X-Ryl669 which enable the support of upcoming ESP-IDF version. Actually, it is a preliminary support and it doesn’t support ULP.

You can find the blink example [here](https://github.com/jonathandreyer/platformio-esp32s3-blink).

Best regards,

Jonathan Dreyer

ralphrmartin, 2022-03-26:
It would also be nice to have support for the muselab nanoESP32-s3 board
[https://www.tindie.com/products/johnnywu/nanoesp32-s3-development-board/]
which I guess would only be a little extra effort once esp32-s3 tooling is in place.

jonathandreyer, 2022-04-23:
Hi everyone,

I have updated the platform repository with ESP-IDF v4.4.1. I have tested it with the board mentioned in the title of the issue. To use it, it is only necessary to replace the platform by:
```
platform = https://github.com/jonathandreyer/platform-espressif32.git#espidf-441
```

Best regards,

Jonathan Dreyer

jonathandreyer, 2022-04-28:
Hi everyone,

I did a quick update to inform you that the debug (by the internal JTAG interface) is also usable. The `upload_protocol` and `debug_tool` are `esp_usb_jtag`.

Best regards,

Jonathan Dreyer

code-is-art, 2022-05-14:
This json from espressif has been updated to include the esp32-s3 dev board for arduino-esp32. I've been able to compile using the arduino ide for my esp32-s3 dev board. Not sure how to make it work for pio.
```
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
```

closed 2022-05-16 by valeros

kevinhikaruevans, 2022-06-01:
It looks like only Arduino is supported. Will there be support for the espidf?

Jason2866, 2022-06-02:
@kevinhikaruevans #710 is still open. Until it is official released you can use my fork which supports Arduino core 2.0.3 and IDF 4.4.1. Using Arduino as Component of IDF does work too.
```
https://github.com/Jason2866/platform-espressif32
```

commit 9f9f39d8 (2022-05-16): Add support for Arduino core v2.0.3

Resolve #799, resolves #762, resolves #674