platformio/platform-espressif32 #675: esp32_exception_decoder wrong result

smarq8, 2021-11-21:
using "monitor_filters = esp32_exception_decoder" return incorrect result in file and fileLine
for example:
```
0x4013c084:0x3ffced90 in TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at lib\TFT_eSPI/TFT_eSPI.cpp:3403
should be
0x4013c084: TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 257
```
For comparison [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.1.0) tool for Arduino seems to be correct at decoding, and I found there was some trouble with decoding fileLine mentioned [here](https://github.com/me-no-dev/EspExceptionDecoder/commit/a78672da204151cc93979a96ed9f89139a73893f). I hope it will help somehow to solve problem.

[image]

[image]

valeros, 2021-11-22:
Hi @smarq8 ! I believe it's a peculiarity of the `TFT_eSPI` library. They [use the preprocessor](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L19-L29) to combine several source files into one compilation unit, so the backtrace seems to be technically correct, but not really informative. Although it's interesting how the tool for Arduino is capable of handling case. 

@Tasssadar any chance you could take a look at this?

smarq8, 2021-11-22:
After further investigation I probably found interesting trace. In very short, it seems "xtensa-esp32-elf-addr2line.exe" might be outdated. Just read carefuly descrition of each code block and analyze for example fileLine of "TFT_eSPI::pushSwapBytePixels"
decode using esp32_exception_decoder from platformio
```
Backtrace: 0x4013e600:0x3ffcec90 0x4013e835:0x3ffcecf0 0x401401cc:0x3ffced10 0x401366c4:0x3ffced40 0x400fb533:0x3ffced60 0x400faad6:0x3ffced90 0x400facc1:0x3ffcef90 0x400fb5f3:0x3ffcefe0 0x400fb643:0x3ffcf060 0x400d47fc:0x3ffcf0a0 0x400d4931:0x3ffcf0e0 0x400d4add:0x3ffcf120 0x400e79e9:0x3ffcf140 0x400d4d46:0x3ffcf180 0x400fe41d:0x3ffcf1a0 0x4008fc0a:0x3ffcf1c0  
  #0  0x4013e600:0x3ffcec90 in TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at lib\TFT_eSPI/TFT_eSPI.cpp:3403
  #1  0x4013e835:0x3ffcecf0 in TFT_eSPI::pushPixels(void const*, unsigned int) at lib\TFT_eSPI/TFT_eSPI.cpp:3403
  #2  0x401401cc:0x3ffced10 in TFT_eSPI::pushColors(unsigned short*, unsigned int, bool) at lib\TFT_eSPI/TFT_eSPI.cpp:3403
  #3  0x401366c4:0x3ffced40 in hagl_hal_push_area at lib\hagl_hal_display_TFT_eSPI/hagl_hal_display_TFT_eSPI.cpp:38
  #4  0x400fb533:0x3ffced60 in hagl_push_area at lib\hagl\src/hagl.c:417
  #5  0x400faad6:0x3ffced90 in hagl_put_char at lib\hagl_hal_font_adafruit/hagl_hal_font_adafruit.c:59
  #6  0x400facc1:0x3ffcef90 in hagl_hal_draw_string_XYCBS at lib\hagl_hal_font_adafruit/hagl_hal_font_adafruit.c:59
  #7  0x400fb5f3:0x3ffcefe0 in hagl_vprintf_XYSCB at lib\hagl\src/hagl.c:417
  #8  0x400fb643:0x3ffcf060 in hagl_printf_XYSCB at lib\hagl\src/hagl.c:417
  #9  0x400d47fc:0x3ffcf0a0 in dualAxisProtractor::draw(int, int, int, int, vect3) at src/dualAxisProtractor.cpp:142
  #10 0x400d4931:0x3ffcf0e0 in dualAxis_draw(int, int, int, int, void*) at src/dualAxisProtractor.cpp:75
  #11 0x400d4add:0x3ffcf120 in _lv_directDraw_handler at src/lvgl_ex.cpp:22 (discriminator 2)
  #12 0x400e79e9:0x3ffcf140 in lv_timer_exec at lib\lvgl\src\misc/lv_timer.c:226
      (inlined by) lv_timer_handler at lib\lvgl\src\misc/lv_timer.c:106
  #13 0x400d4d46:0x3ffcf180 in loop() at src/main.cpp:98
  #14 0x400fe41d:0x3ffcf1a0 in loopTask(void*) at C:\Users\xXx\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #15 0x4008fc0a:0x3ffcf1c0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1) 
```

decode using Arduino tool "EspExceptionDecoder-1.0.6" - I tested all versions from 1.0.3-1.0.6 with same result
```
Decoding 22 results
0x4013e600: TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 257
0x4013e600: TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 257
0x4013e624: TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 260 (discriminator 2)
0x4013e600: TFT_eSPI::pushSwapBytePixels(void const*, unsigned int) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 257
0x4013e835: TFT_eSPI::pushPixels(void const*, unsigned int) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI\Processors/TFT_eSPI_ESP32.c line 324
0x401401cc: TFT_eSPI::pushColors(unsigned short*, unsigned int, bool) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\TFT_eSPI/TFT_eSPI.cpp line 3103
0x401366c4: hagl_hal_push_area at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl_hal_display_TFT_eSPI/hagl_hal_display_TFT_eSPI.cpp line 38
0x400fb533: hagl_push_area at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl\src/hagl.c line 1108
0x400faad6: hagl_put_char at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl_hal_font_adafruit/hagl_hal_font_adafruit.c line 192
0x400facc1: hagl_hal_draw_string_XYCBS at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl_hal_font_adafruit/hagl_hal_font_adafruit.c line 354
0x400fb5f3: hagl_vprintf_XYSCB at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl\src/hagl.c line 1182
0x400fb643: hagl_printf_XYSCB at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\hagl\src/hagl.c line 1154
0x400d47fc: dualAxisProtractor::draw(int, int, int, int, vect3) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/src/dualAxisProtractor.cpp line 142
0x400d4931: dualAxis_draw(int, int, int, int, void*) at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/src/dualAxisProtractor.cpp line 75
0x400d4add: _lv_directDraw_handler at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/src/lvgl_ex.cpp line 45
0x400e79e9: lv_timer_exec at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\lvgl\src\misc/lv_timer.c line 315
:  (inlined by) lv_timer_handler at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\lvgl\src\misc/lv_timer.c line 106
0x400d4d46: loop() at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/src/main.cpp line 98
0x400fe41d: loopTask(void*) at C:\Users\xXx\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp line 23
[... cut ...]
0x400e79e9: lv_timer_exec at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\lvgl\src\misc/lv_timer.c:226
 (inlined by) lv_timer_handler at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/lib\lvgl\src\misc/lv_timer.c:106
0x400d4d46: loop() at C:\Users\xXx\Documents\PlatformIO\Projects\myProtractor/src/main.cpp:98
0x400fe41d: loopTask(void*) at C:\Users\xXx\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
0x4008fc0a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

closed 2022-09-20 by stale[bot]