platformio/platform-espressif32 #680: Adafruit BusIO 1.9.7 breaks when compiling with VSCode+PlatformIO, they say it's because of not havine the latest framework

han-fastolfe, 2021-11-25:
The error looks like this:

```
Processing wemos_d1_mini32 (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 HUB75 LED MATRIX PANEL DMA Display> 2.0.6
|   |-- <Adafruit GFX Library> 1.10.12
|   |   |-- <Adafruit BusIO> 1.9.7
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.4.0
|   |   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.9.7
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Adafruit GFX Library> 1.10.12
|   |-- <Adafruit BusIO> 1.9.7
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/FastLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/bitswap.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorpalettes.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorutils.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/hsv2rgb.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/lib8tion.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/noise.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms/esp/32/clockless_rmt_esp32.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/power_mgt.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/wiring.cpp.o
Compiling .pio/build/wemos_d1_mini32/src/main.cpp.o
Generating partitions .pio/build/wemos_d1_mini32/partitions.bin
Compiling .pio/build/wemos_d1_mini32/lib2ea/Wire/Wire.cpp.o
Compiling .pio/build/wemos_d1_mini32/libcd9/SPI/SPI.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
Archiving .pio/build/wemos_d1_mini32/libcd9/libSPI.a
Indexing .pio/build/wemos_d1_mini32/libcd9/libSPI.a
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::interruptHandler(void*)':
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:318:10: warning: unused variable 'stuff_to_do' [-Wunused-variable]
     bool stuff_to_do = false;
          ^
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GFX.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GrayOLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_SPITFT.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/glcdfont.c.o
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp: In member function 'void Adafruit_I2CDevice::end()':
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp:44:10: error: 'class TwoWire' has no member named 'end'
   _wire->end();
          ^
Compiling .pio/build/wemos_d1_mini32/lib5ba/ESP32 HUB75 LED MATRIX PANEL DMA Display/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o
*** [.pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o] Error 1
========================================================== [FAILED] Took 2.28 seconds ==========================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Reported [here](https://github.com/adafruit/Adafruit_BusIO/issues/69#issuecomment-979419026) where they say the problem is with not having the latest ESP32 Arduino framework.

han-fastolfe, 2021-11-26:
```
cory@RedBarchetta[10:40:51]~/Documents/PlatformIO/Projects/RGB-Matrix-1$ pio platform update
Platform espressif32
--------
Updating platformio/espressif32               3.4.0                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32ulp        1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating espressif/toolchain-xtensa-esp32     8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating espressif/toolchain-xtensa-esp32s2   8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating espressif/toolchain-riscv32-esp      8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10006.210326 @ ~3.10006.0        [Up-to-date]
Updating platformio/framework-espidf          3.40301.0 @ ~3.40301.0             [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-ninja                1.7.1 @ ^1.7.0                     [Up-to-date]
```

That third component in the version number for `platformio/framework-arduinoespressif32`,  `3.10006.210326`, looks an awful lot like a date. If it is, that means that `platformio/framework-arduinoespressif32` is still Espressif's 1.0.6 version and has yet to update to the 2.0.0 version which came out in August.

That `2.0.0` version does include the `end` method, which is not there in `1.0.6` version PIO uses. Since `2.0.0` is 3 months old, the PIO frameworks should be upgraded to use it ASAP.

solhuebner, 2021-11-27:
PlatformIO just implemented IDF 4.3.1 (https://github.com/platformio/platform-espressif32/releases/tag/v3.4.0) but Arduino-ESP32 2.0.1 is based on 4.4 (https://github.com/espressif/arduino-esp32/releases/tag/2.0.1).

So you would either need to wait or do the following in your platformio.ini:
change platform to: `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`
and add:
```
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
```
This is how mine looks like:
[image]

Ideally run a "Clean All" afterwards:
[image]

han-fastolfe, 2021-11-27:
@solhuebner Thanks for that fix, it worked for me. Where would I find that info in Espressif official sources to know how to stay up to date and help others in the future?

solhuebner, 2021-11-30:
@han-fastolfe that is difficult as PlatformIO is maintaining their own package as they do Q&A first (big user base). With the method I described you can influence the version but with all bleeding edge things there might be issues along the way. The reason I played around with it is that I want ESP32-S3 support as soon as they release it but that will probably not before next year as even the latest IDF has not everything covered yet ;)

Jason2866, 2021-12-28:
Try this built with Arduino Core v2.0.2 and IDF4.4
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
```

valeros, 2022-05-04:
Should be fixed in the latest version of the dev-platform. Please reopen if the problem persists.

closed 2022-05-04 by valeros