platformio/platform-espressif32 #682: WDT espressif example doesnt work in platformio

lossonfire, 2021-11-29:
This example doesnt work on platformio but works ok in arduino IDE V1.8.16
I'm using Linux Mint Debian Edition 4 and the lastest version of Vs Code and Platformio.

https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/WatchdogTimer/WatchdogTimer.ino

[image]

mrengineer7777, 2021-11-30:
It might need this at the top of the file:
`#include <Arduino.h>`

lossonfire, 2021-12-01:
> It might need this at the top of the file: `#include <Arduino.h>`

Thanks friend, but was included, this its the code:

`#include <Arduino.h>

//https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/WatchdogTimer/WatchdogTimer.ino

#include "esp_system.h"

const int button = 0;         //gpio to use to trigger delay
const int wdtTimeout = 5000;  //time in ms to trigger the watchdog
hw_timer_t *timer = NULL;

void ARDUINO_ISR_ATTR resetModule() {
  ets_printf("reinicio por WDT\n");
  esp_restart();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("running setup");

  pinMode(button, INPUT_PULLUP);                    //init control pin
  timer = timerBegin(0, 80, true);                  //timer 0, div 80
  timerAttachInterrupt(timer, &resetModule, true);  //attach callback
  timerAlarmWrite(timer, wdtTimeout * 1000, false); //set time in us
  timerAlarmEnable(timer);                          //enable interrupt
}

void loop() {
  Serial.println("running main loop");

  timerWrite(timer, 0); //reset timer (feed watchdog)
  long loopTime = millis();
  //while button is pressed, delay up to 3 seconds to trigger the timer
  while (!digitalRead(button)) {
    Serial.println("button pressed");
    delay(500);
  }
  delay(1000); //simulate work
  loopTime = millis() - loopTime;
  
  Serial.print("loop time is = ");
  Serial.println(loopTime); //should be under 3000
}`

mrengineer7777, 2021-12-01:
Issue reproduced.  Solution below.

I created a new PlatformIO project with Arduino framework.  Replaced contents of main.cpp with Github code linked above.  Compile fails.  Doesn't recognize "ARDUINO_ISR_ATTR".
Renamed main.cpp -> main.ino.  Intellisense doesn't like it, but normally PlatformIO will compile it.  Compile still fails.  Doesn't recognize "ARDUINO_ISR_ATTR".

Searched for "ARDUINO_ISR_ATTR" in all installed PlatformIO packages.  No results.

Renamed main.ino -> main.cpp.
Using VSCode Intelllisense on timerAttachInterrupt()  (Ctrl+click in Windows):
Parameter two is void (*fn)(void).

Changed resetModule() to
`void resetModule() {
  ets_printf("reboot\n");
  esp_restart();
}`

Compile succeeds.  Gotta love Intellisense.

valeros, 2022-05-04:
The sketch above compiles without errors with the latest version of the platform. Feel free to reopen the issue if the problem persists.

closed 2022-05-04 by valeros