platformio/platform-espressif32 #684: Could not install package 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1' for 'linux_x86_64' system

SebiTimeWaster, 2021-12-01:
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
Error: Could not install package 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1' for 'linux_x86_64' system
The terminal process "platformio 'run'" terminated with exit code: 1.
```

it would be nice if the tool would tell me WHY it can't install it...

also what does that mean:

```
Updating platformio/tool-scons                4.40200.0 @ ~4.40200.0             [Incompatible 4.40300.1]
```

ivankravets, 2021-12-01:
Sorry, fixed on our side. Please try again.

closed 2021-12-01 by ivankravets

SebiTimeWaster, 2021-12-01:
thanks for the fast reply, it starts to compile now, but i run into multiple of these errors:

```
In file included from include/sd-xxx.h:8,
                 from src/sd-xxx.cpp:1:
/home/xxx/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from /home/xxx/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from include/sd-xxx.h:8,
                 from src/sd-xxx.cpp:1:
/home/xxx/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
```

SebiTimeWaster, 2021-12-02:
i solved it by putting
```
#ifdef __cplusplus
extern "C" {
#endif
#include <dirent.h>
#ifdef __cplusplus
}
#endif
```
around the dirent include.