platformio/platform-espressif32 #685: examples/espidf-arduino-blink build fails: unknown type name '_lock_t'

skiphansen, 2021-12-11:
I followed the directions in platform-espressif32/examples/espidf-arduino-blink/README.md exactly, but the build failed with  ````error: unknown type name '_lock_t'```` in lots of places.

Instead of using the zip file I also tired cloning  https://github.com/platformio/platform-espressif32.git directly but the example failed the same way in both the master and develop branches.

The examples/arduino-blink builds fine.

I'm new to platformio and I'm trying to build a project that uses classic bluetooth which appears to need a tweak using menuconfig to build.  Researching that issue lead me to the espidf-arduino-blink.

Here's my [build.log](https://github.com/platformio/platform-espressif32/files/7696088/build.log).

skiphansen, 2021-12-11:
I looks like the wrong lock.h is getting included.  I've build quite a few projects since I first installed platform IO a week or so ago for several different boards.

I  just created a temporary user and installed platformio from scratch and then tried the example and it built without a problem.

Earlier I thought I had caught platformio using the main repo of a library when platformio.ini has specified a particular branch, but I didn't chase that down.

There's still an issue somewhere, but it isn't specific to this example.

maxgerhardt, 2021-12-18:
Sounds simliar to https://github.com/platformio/platform-espressif32/issues/646#issuecomment-948358453, does the same workaround work?

skiphansen, 2021-12-18:
I don't know.  When I discovered that a virgin install didn't have the issue I wiped my .platformio/packages directory and that fixed it.  I think there's still an issue with a builds not using the correct headers.  I don't know enough about platformio to chase it further.

Jason2866, 2021-12-28:
I builded a actual framework. Arduino Core v2.0.2 and IDF v4.4.
All included examples in this builded framework do work
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
```

60pfennig, 2022-01-20:
@skiphansen Did you find a solution? I have the same problem.

maxgerhardt, 2022-01-20:
He already said

>wiped my .platformio/packages directory and that fixed it. 

Same as [this person](https://community.platformio.org/t/arduino-esp-idf-newly-broken/25552).

dumarjo, 2022-02-11:
@Jason2866 I tried your build to be able to use the arduino,espidf sample (https://github.com/tasmota/platform-espressif32/tree/Arduino2022/IDF44/examples/espidf-arduino-wifiscan) , and I have an error about the crt_bundle file not found...

```ini
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.2/platform-espressif32-2.0.2.zip
```

```powershell
Generating assembly for certificate bundle...
CMake Error at C:/Users/xyz/.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
  file failed to open for reading (No such file or directory):

    D:\project\.pio\build\esp32dev\x509_crt_bundle
```
It's look like a problem with python3 be more strict on validating the certificate bundles. Do you have an idea how to fix that ?
looks related to https://community.platformio.org/t/problems-building-with-esp-idf-latest-version-4-3-0/23758/5 but i'm not 100% sure. 

Regards,

closed 2022-09-20 by stale[bot]