platformio/platform-espressif32 #688: add option for xRingbufferCreateStatic to use PSRAM

bhupiister, 2021-12-14:
Want to create a static ring buffer to use PSRAM on esp32. With the inbuilt FreeRTOS, there is no option to create xRingbufferCreateStatic.
```
  //Allocate memory for ring buffer data structure
  StaticRingbuffer_t *buffer_struct = (StaticRingbuffer_t *)heap_caps_malloc(sizeof(StaticRingbuffer_t), MALLOC_CAP_SPIRAM);
  //Allocate memory for ring buffer's data storage area
  uint8_t *buffer_storage = (uint8_t *)heap_caps_malloc(sizeof(uint8_t) * buffSizeInBytes, MALLOC_CAP_SPIRAM);
  //Set your buffer type
  RingbufferType_t buffer_type = RINGBUF_TYPE_NOSPLIT;
  buf_handle = xRingbufferCreateStatic(buffSizeInBytes, buffer_type, buffer_storage, buffer_struct);
```

valeros, 2021-12-15:
Hi @bhupiister , This repository is not the right place to ask such generic questions,  please forward it to the official ESP-IDF repository https://github.com/espressif/esp-idf

closed 2021-12-15 by valeros

daz, 2023-04-19:
Is there an updated way to do this? The `feature/arduino-upstream` branch is currently 2 years behind the dev branch, but I haven’t found any other way to get of the error:

` error: implicit declaration of function 'xQueueCreateStatic';`

daz, 2023-05-07:
My fix was in `FreeRTOSConfig.h` specifically setting`configSUPPORT_STATIC_ALLOCATION`

bhupiister, 2023-05-07:
@daz you must be using esp idf to make changes to freertos.