platformio/platform-espressif32 #690: Cannot build with feature/arduino-upstream

han-fastolfe, 2021-12-19:
Changing from

```
[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
framework = arduino
lib_extra_dirs = ~/Development/Arduino/sketchbook/libraries
```

to 

```
[env:wemos_d1_mini32]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
board = wemos_d1_mini32
framework = arduino
lib_extra_dirs = ~/Development/Arduino/sketchbook/libraries
```

then doing a "Clean All" will no longer build in a brand new project.

Output:

```
> Executing task in folder PNGdec Test: platformio run --environment wemos_d1_mini32 <

Processing wemos_d1_mini32 (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: wemos_d1_mini32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.d218e58 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa-esp32 8.4.0+2021r2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/wemos_d1_mini32/src/main.cpp.o
Generating partitions .pio/build/wemos_d1_mini32/partitions.bin
Archiving .pio/build/wemos_d1_mini32/libFrameworkArduinoVariant.a
Indexing .pio/build/wemos_d1_mini32/libFrameworkArduinoVariant.a
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/Print.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/USB.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/USBMSC.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/WString.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/base64.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-i2c-slave.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-ledc.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-matrix.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/main.cpp.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/wemos_d1_mini32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/wemos_d1_mini32/libFrameworkArduino.a
Indexing .pio/build/wemos_d1_mini32/libFrameworkArduino.a
Linking .pio/build/wemos_d1_mini32/firmware.elf
xtensa-esp32-elf-g++: error: Test.map: No such file or directory
*** [.pio/build/wemos_d1_mini32/firmware.elf] Error 1
================================================= [FAILED] Took 1.48 seconds =================================================
The terminal process "platformio 'run', '--environment', 'wemos_d1_mini32'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Unaltered main.cpp from project creation:

```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

I made the switch to the feature branch because PlatformIO's usual stuff is not currently up to date with the latest from Espressif which causes issues like #680.

Jason2866, 2021-12-19:
This works
```
[env:esp32dev]
platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
                              platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip

```

Jason2866, 2021-12-28:
Or use this. Arduino v.2.0.2 and IDF v.4,4 working
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
```

valeros, 2022-04-18:
Should be fixed in https://github.com/espressif/arduino-esp32/pull/6464

closed 2022-04-18 by valeros