platformio/platform-espressif32 #695: Help with ESP-IDF v4.4 linker logic

maxgerhardt, 2021-12-27:
Per https://community.platformio.org/t/platformio-support-for-esp32-coredump/25141 I'm trying to get PlatformIO to build with a ESP-IDF v4.4 pre-release version (https://github.com/espressif/esp-idf/tree/release/v4.4). 

I've forked ESP-IDF v4.4, added the needed package.json and referenced this as the source for `framework-espidf´ in the `platform_packages` along with a newer XTensa-ESP32 compiler version (2021r2). This gets a menuconfig with the new version going, however, building fails pretty quickly. 

ESP-IDF v4.4 changed the linker script logic (-.-*). Where there previously was a `esp32_out.ld` generated from a single `.ld.in` file, there now is a `sections.ld.in` and `memory.ld.in` that need to be taken into account, which reside in components which names were changed compared to the previous ones. 

I have adapted the build logic in https://github.com/maxgerhardt/platform-espressif32/commit/6513b99dbf1a201d8a688bac43b3d91afb68145b, and this does handle `sections.ld` and `memory.ld` correctly, but linking still fails due to 1. flags in the linker command that are missing with respect to a reference ESP-IDF build and 2. the `-L` flags have the path of the builder script in front of the actual path, which screws up detection of the linker paths where needed files are. This is inexplicable to me.

Current linker command with the above builder script:

>`xtensa-esp32-elf-gcc -o .pio\build\esp32dev\bootloader.elf -Wl,--Map=C:/Users/Max/temp/esp32_dump/.pio/build/esp32dev/bootloader/bootloader.map -Wl,--cref -Wl,--gc-sections -Wno-frame-address -fno-lto -fno-rtti -mlongcalls -u __assert_func -T esp32.peripherals.ld -u abort -u __ubsan_include -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-funcs.ld -T bootloader.ld -T bootloader.rom.ld -u bootloader_hooks_include -Wl,--start-group .pio\build\esp32dev\bootloader\esp-idf\hal\libhal.a .pio\build\esp32dev\bootloader\esp-idf\main\libmain.a .pio\build\esp32dev\bootloader\esp-idf\bootloader_support\libbootloader_support.a .pio\build\esp32dev\bootloader\esp-idf\efuse\libefuse.a .pio\build\esp32dev\bootloader\esp-idf\esp_common\libesp_common.a .pio\build\esp32dev\bootloader\esp-idf\esp_hw_support\libesp_hw_support.a .pio\build\esp32dev\bootloader\esp-idf\esp_rom\libesp_rom.a .pio\build\esp32dev\bootloader\esp-idf\esp_system\libesp_system.a .pio\build\esp32dev\bootloader\esp-idf\log\liblog.a .pio\build\esp32dev\bootloader\esp-idf\micro-ecc\libmicro-ecc.a .pio\build\esp32dev\bootloader\esp-idf\soc\libsoc.a .pio\build\esp32dev\bootloader\esp-idf\spi_flash\libspi_flash.a .pio\build\esp32dev\bootloader\esp-idf\xtensa\libxtensa.a -L.pio\build\esp32dev -LC:\Users\Max\.platformio\platforms\espressif32@src-fc18404556a6086e27708214e4501bbd\builder\frameworks\"C:\Users\Max\.platformio\packages\framework-espidf\components\soc\esp32\ld" -LC:\Users\Max\.platformio\packages\framework-espidf\components\xtensa\esp32 -LC:\Users\Max\.platformio\platforms\espressif32@src-fc18404556a6086e27708214e4501bbd\builder\frameworks\"C:\Users\Max\.platformio\packages\framework-espidf\components\esp_rom\esp32\ld" -LC:\Users\Max\.platformio\platforms\espressif32@src-fc18404556a6086e27708214e4501bbd\builder\frameworks\"C:\Users\Max\.platformio\packages\framework-espidf\components\bootloader\subproject\main\ld\esp32" -Wl,--end-group -lxt_hal`

Notice flags like

>`-LC:\Users\Max\.platformio\platforms\espressif32@src-fc18404556a6086e27708214e4501bbd\builder\frameworks\"C:\Users\Max\.platformio\packages\framework-espidf\components\bootloader\subproject\main\ld\esp32" `

Which should be just `-L"C:\Users\Max\.platformio\packages\framework-espidf\components\bootloader\subproject\main\ld\esp32"`. 

The reference project is at https://github.com/maxgerhardt/pio-espidf-4.4-test.

The reference ESP-IDF verbose build output is at https://github.com/maxgerhardt/pio-espidf-4.4-test/blob/main/build.log.

Jason2866, 2021-12-27:
Have you seen this https://github.com/platformio/platform-espressif32/pull/693

maxgerhardt, 2021-12-27:
Thanks @Jason2866, this helped me get a successful build through https://github.com/maxgerhardt/platform-espressif32/commit/5a537a4074650858a279f3620a976f54533d7531, now onwards to testing...

maxgerhardt, 2021-12-27:
Yep, that works.

closed 2021-12-27 by maxgerhardt