platformio/platform-espressif32 #699: Toolchain-xtensa-esp32s2 for linux_aarch64

maedball, 2022-01-04:
Hiho,

i would be super helpful for all Homeassistant Blue users with ESP home to have the toolchain-xtensa-esp32s2 available in linux_aarch64.

Would it be possible to make this available?

Thanks

homonto, 2022-01-04:
+1 from me - I have a bunch of ESP32-S2 and I cannot use them with ESPHome that is on Raspberry Pi

hmax42, 2022-01-05:
the same problem affects esp32-c3

DonBosserez, 2022-01-05:
+1 for me - also running HASS on Odroid hardware.

GNAKACH, 2022-01-12:
Hi Same problem for me; Same configuration: Homme assistant running on a raspberry Pi 4 with Home assistant add on.
It is really a fantastic tool.
alas, I cannot use my ESP32-S2 as the platform is missing:(
I hope the implementation arrives soon ;)

maedball, 2022-01-12:
> > Hi Same problem for me; Same configuration: Homme assistant running on a raspberry Pi 4 with Home assistant add on.
> > It is really a fantastic tool.
> > alas, I cannot use my ESP32-S2 as the platform is missing:(
> > I hope the implementation arrives soon ;)
> 
> I "solved" this problem by programming in Arduino for the boards that are not supported But pain is there, and instead of native ESPHome protocol I am using MQTT of course At least I can learn more about C and MQTT while waiting for ESPHome supporting new boards on arm64 ;-)

You can also use ESPHOME installed on a normal x64 linux as a workaround.

homonto, 2022-01-12:
> You can also use ESPHOME installed on a normal x64 linux as a workaround.

I thought about it but I have at home only arm based computers: RPi and Macbooks
So not even VM will help

rbray89, 2022-01-18:
Same. Been running on a docker instance on an x86 -64 machine to get around this. With RPI4 being aarch64, you'd think this would be more urgentl.

Jason2866, 2022-01-23:
Toolchains are directly provided (since a while) from espressif to the Platformio Registry. The feature request is better placed in the corresponding espressif github.
https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/compatibility
https://github.com/espressif/crosstool-NG.git

homonto, 2022-01-23:
> Toolchains are directly provided (since a while) from espressif to the Platformio Registry. The feature request is better placed in the corresponding espressif github.
> https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/compatibility
> https://github.com/espressif/crosstool-NG.git

you are saying: there is not toolchain for arm64? so how is this working on Arduino?

Jason2866, 2022-01-23:
I just say if you want to use a toolchain from espressif for platformio you have to ask espressif. Not more not less. Idk if there is a toolchain or not.

maedball, 2022-01-25:
https://github.com/espressif/crosstool-NG/issues/21

Opened ;-)

quentinmit, 2022-02-28:
As a workaround you can invoke platformio or esphome with `linux32`, which will cause platformio to think it's running in 32-bit mode:

```
# If not already enabled on your system
sudo dpkg --add-architecture armel
sudo apt update
sudo apt install libc6:armel libstdc++6:armel
# Now build hello world
mkdir /tmp/test
cd /tmp/test
pio project init --board esp32-c3-devkitm-1
linux32 pio run
```

Jason2866, 2022-04-14:
I have builded a aarch64 toolchain set for Linux aarch64 useable with Platformio
https://github.com/Jason2866/crosstool-NG/releases

platima, 2022-05-03:
> I have builded a aarch64 toolchain set for Linux aarch64 useable with Platformio https://github.com/Jason2866/crosstool-NG/releases

You are an amazing person <3

GNAKACH, 2022-05-04:
Brilliant! A million thanks!

platima, 2022-05-04:
Note had to set 
```
  platformio_options:
    board_build.flash_mode: dio
```

Jason2866, 2022-05-04:
@platima Option is in no way related to the toolchain. "Just" the board

platima, 2022-05-09:
> @platima Option is in no way related to the toolchain. "Just" the board

Oh, would not all esp32s2's be flashed that same way then? I thought it was a chip specification, not board!

Cheers

Jason2866, 2022-05-09:
No, since it is a option for the flash chip and how it is connected to the esp32s2
There are different options for the flash chip and how it is connected.

platima, 2022-05-09:
Good to know! Thanks mate

Jason2866, 2022-08-10:
Supported. Use latest Platform-espressif32 release.

platima, 2022-08-13:
Sweet! Thansk