platformio/platform-espressif32 #709: Unable to mount SD using feature/arduino-upstream

idorurez, 2022-01-23:
Apologies, I am not an engineer by trade, but I can navigate code.

I have been unable to mount my SD card using the following platformio.ini config:

```

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
upload_port = COM5
monitor_speed = 115200
framework = arduino
platform_packages = 
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
	toolchain-xtensa32@~2.80400.0
lib_deps = 
	bblanchon/ArduinoJson@^6.18.5
	plerup/EspSoftwareSerial@^6.15.1
	adafruit/Adafruit BME280 Library@^2.2.1
	bodmer/JPEGDecoder@^1.8.1
	claws/BH1750@^1.2.0
	bodmer/TFT_eSPI@^2.4.25
```

**Otherwise, I can mount the SD card just fine if I don't use the upstream and any of the platform packages mentioned above**. This was a hold out from a previous project that required this. Removing it allows me to mount the SD card with no issue.

I had originally thought it was my pick of either VSPI or HSPI chip select pins (and other peripherals connected), but starting a fresh new project with no other components except for the SD card, I was still getting the same mount error until I reverted it back to:
```

[env:esp32dev]
; platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform = espressif32
board = esp32dev
upload_port = COM5
monitor_speed = 115200
framework = arduino
; platform_packages = 
;	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
;	toolchain-xtensa32@~2.80400.0
lib_deps = 
	bblanchon/ArduinoJson@^6.18.5
	plerup/EspSoftwareSerial@^6.15.1
	adafruit/Adafruit BME280 Library@^2.2.1
	bodmer/JPEGDecoder@^1.8.1
	claws/BH1750@^1.2.0
	bodmer/TFT_eSPI@^2.4.25
```

Figured I'd let someone know. Thank you!

Jason2866, 2022-01-23:
Platformio github issues is the wrong place for this. It is NOT a Platformio issue.
Your setup is not correct. You can not use old toolchains with latest Arduino.
SD Card mount may not work, since there was (is?) a issue in latest Arduino code.

idorurez, 2022-01-23:
Ok

closed 2022-01-23 by idorurez