platformio/platform-espressif32 #718: ESP32, arduino and custom boolader

Koxx3, 2022-02-03:
Hello,
I have an arduino project for ESP32.
I'd like to enable flash encryption.
I moved to "arduino, espidf". It's building.
Now, I'd like to rebuild the bootloader to enable flash encryption and use this custom bootloader.
Is there any possibility ?

Jason2866, 2022-02-04:
Not really a Platformio issue nor a question. Afaik Arduino does not support flash encryption. Building bootloader with support for needs recompiling the libs from the Arduino framework with the coresponding settings in sdkconfig done.
See https://github.com/espressif/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32#L69-L75
After you have builded the framework you can use with platformio

Koxx3, 2022-02-04:
I use "arduino, espidf"
I've configure sdkconfig with secure boot & flash encryption.

But now, on building, PIO fails with the following message : 
*** [.pio\build\xxx\.pio\build\xxx\signature_verification_key.bin.o] Source `.pio\build\xxx\signature_verification_key.bin.S' not found, needed by target `.pio\build\xxx\.pio\build\xxx\signature_verification_key.bin.o'.

so, I think there is something PIO related. but I really don't see how to solve that.

Jason2866, 2022-02-04:
You are right PIO does not support.

viniciusdesa, 2022-09-04:
Is there an estimative to this feature?