platformio/platform-espressif32 #724: Bad CPU type in executable - new M1 Pro

84ace, 2022-02-08:
Hi, I have a brand new Apple M1 Pro Macbook with VSCode and Platformio Installed. Just added my project and went to build it and this is the result:
```
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.50200.0
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa32 @ 2.50200.97 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0+sha.d93c9ef) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <AsyncElegantOTA> 2.2.6
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <ESPAsyncWebServer-esphome> 2.1.0
|   |   |-- <AsyncTCP-esphome> 1.2.2
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|-- <ESP_DoubleResetDetector> 1.2.1
|   |-- <LittleFS_esp32> 1.0.6
|   |   |-- <FS> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32TimerInterrupt> 1.5.0
|-- <SparkFun I2C GPS Reading and Control> 1.0.4
|   |-- <Wire> 1.0.1
|-- <TinyGPSPlus> 1.0.3
|-- <LittleFS_esp32> 1.0.6
|   |-- <FS> 1.0
|-- <Uptime Library> 1.0.0
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Preferences> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <SimpleTimer> 1.0.0
|-- <Wire> 1.0.1
Building in release mode
Just tried to upload files to LittleFS partition...
Compiling .pio/build/esp32dev/src/main.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib236/AsyncTCP-esphome/AsyncTCP.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/libd80/FS/FS.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/libd80/FS/vfs_api.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib14e/WiFi/ETH.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib14e/WiFi/WiFi.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib14e/WiFi/WiFiAP.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib14e/WiFi/WiFiClient.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
Compiling .pio/build/esp32dev/lib14e/WiFi/WiFiGeneric.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
*** [.pio/build/esp32dev/src/main.cpp.o] Error 126
*** [.pio/build/esp32dev/lib236/AsyncTCP-esphome/AsyncTCP.cpp.o] Error 126
*** [.pio/build/esp32dev/libd80/FS/FS.cpp.o] Error 126
*** [.pio/build/esp32dev/libd80/FS/vfs_api.cpp.o] Error 126
*** [.pio/build/esp32dev/lib14e/WiFi/ETH.cpp.o] Error 126
*** [.pio/build/esp32dev/lib14e/WiFi/WiFi.cpp.o] Error 126
*** [.pio/build/esp32dev/lib14e/WiFi/WiFiAP.cpp.o] Error 126
*** [.pio/build/esp32dev/lib14e/WiFi/WiFiClient.cpp.o] Error 126
*** [.pio/build/esp32dev/lib14e/WiFi/WiFiGeneric.cpp.o] Error 126
Compiling .pio/build/esp32dev/lib14e/WiFi/WiFiMulti.cpp.o
sh: /Users/acea/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
*** [.pio/build/esp32dev/lib14e/WiFi/WiFiMulti.cpp.o] Error 126
============================================================================ [FAILED] Took 3.81 seconds ============================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Google isn't being much help. Rosetta isn't installed.

Thanks! Acea.

Jason2866, 2022-02-09:
Without Rosetta it will not work. There is no native Apple silicon support.

84ace, 2022-02-09:
> Without Rosetta it will not work. There is no native Apple silicon support form the used espressif framework. Platformio runs well with Rosetta (use it with Mac M1 Air).

Well, that explains my problem. Thanks for the info.

84ace, 2022-02-09:
Rosetta must be installed as there is no native support for the M1.

closed 2022-02-09 by 84ace

karshPrime, 2024-11-30:
Why is Rosetta2 still a requirement when native support for Apple Silicon has been dropped for ESP-IDF.

Reading the logs and exploring through the install directory, its local download of `clang` that's causing the issue; why can't it use homebrew (/or just system) clang?

```
PIO Core Call Error: "The following files/directories have been created in /Users/zul/Documents/PlatformIO/Projects/BenchX
include - Put project header files here
lib - Put project specific (private) libraries here
src - Put project source files here
platformio.ini - Project Configuration File
Resolving esp32-s3-devkitc-1 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...

UserSideException: Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
OSError: [Errno 86] Bad CPU type in executable: '/Users/zul/.platformio/packages/tool-cmake/bin/cmake':
 File \"/Users/zul/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py\", line 173:
 env.SConscript(\"$BUILD_SCRIPT\")
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py\", line 620:
 return _SConscript(self.fs, *files, **subst_kw)
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py\", line 280:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
 File \"/Users/zul/.platformio/platforms/espressif32/builder/main.py\", line 368:
 target_elf = env.BuildProgram()
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py\", line 252:
 return self.method(*nargs, **kwargs)
 File \"/Users/zul/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py\", line 61:
 env.ProcessProgramDeps()
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py\", line 252:
 return self.method(*nargs, **kwargs)
 File \"/Users/zul/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py\", line 121:
 env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py\", line 252:
 return self.method(*nargs, **kwargs)
 File \"/Users/zul/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py\", line 347:
 SConscript(env.GetFrameworkScript(name), exports=\"env\")
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py\", line 684:
 return method(*args, **kw)
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py\", line 620:
 return _SConscript(self.fs, *files, **subst_kw)
 File \"/Users/zul/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py\", line 280:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
 File \"/Users/zul/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 1506:
 project_codemodel = get_cmake_code_model(
 File \"/Users/zul/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 234:
 run_cmake(src_dir, build_dir, extra_args)
 File \"/Users/zul/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 840:
 run_tool(cmd)
 File \"/Users/zul/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 795:
 result = exec_command(cmd, env=idf_env)
 File \"/Users/zul/.platformio/penv/lib/python3.11/site-packages/platformio/proc.py\", line 114:
 with subprocess.Popen(*args, **kwargs) as p:
 File \"/Users/zul/.platformio/python3/lib/python3.11/subprocess.py\", line 1026:
 self._execute_child(args, executable, preexec_fn, close_fds,
 File \"/Users/zul/.platformio/python3/lib/python3.11/subprocess.py\", line 1950:
 raise child_exception_type(errno_num, err_msg, err_filename)
========================== [FAILED] Took 1.03 seconds =========================="
```