platformio/platform-espressif32 #727: Doesn't work on macOS Monterey 12.3!

VladislavSmolyanoy, 2022-02-10:
Since updating to macOS Monterey 12.3 there have been issues trying to upload code to my ESP32.

macOS Monterey 12.3 officially removes support for python 2.7

I got the Arduino IDE to work by following this guide: https://github.com/espressif/arduino-esp32/issues/4717

Couldn't achieve the same and get the following message:

```
/Users/VladislavS/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/VladislavS/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so: error loading plugin: dlopen(/Users/VladislavS/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/VladislavS/.platformio/packages/toolchain-xtensa32/libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file)
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

Jason2866, 2022-02-11:
Is macOS 12.3 already out?

VladislavSmolyanoy, 2022-02-11:
> Is macOS 12.3 already out?

It is currently in beta with a planned release in two to three weeks.

z90210, 2022-02-16:
If you still can run "python" in Terminal.
My workaround is close your Arduino, execute the follow line in Terminal:

open /Applications/Arduino.app

llamaonaskateboard, 2022-02-17:
This was also reported over at https://github.com/espressif/arduino-esp32/issues/6269

I have this same issue on a x86 MacOS 12.3 Beta (21E5206e) for PlatformIO.

Running otool over liblto_plugin.0.so shows that it explicitly links to osxcross for whatever reason:
```
~/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/libexec/gcc/xtensa-esp32-elf/8.4.0 $ otool -L liblto_plugin.0.so
liblto_plugin.0.so:
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 744.19.0)
	/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib (compatibility version 7.0.0, current version 7.25.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 169.3.0)
```
Theoretically the missing library should have always made it fail so it seems like something in MacOS 12.3 is triggering the lto plugin to be used when it wasn't previously.

A workaround that seems to work fine for me (x86 only so won't work for M1/arm64) is installing gcc from Homebrew, creating the osxcross path and copying libstdc++.6.dylib:
```
brew install gcc
sudo mkdir -p /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/
sudo cp /usr/local/Cellar/gcc/11.2.0_3/lib/gcc/11/libstdc++.6.dylib /opt/osxcross/target/x86_64-apple-darwin12/lib/libstdc++.6.dylib
```
I should also add that libstdc++.6.dylib didn't already exist anywhere else on the filesystem.

franzel61, 2022-02-17:
Thanks it seems it works for me ..... back in game :-)

llamaonaskateboard, 2022-02-17:
`platformio-build-esp32.py` from the source repo of framework-arduinoespressif32 at https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32.py#L83 has a lot of LINKFLAG options set including `-fno-lto`, however the latest version (3.10006.210326) from the PlatformIO registry has significantly less options and does not have `-fno-lto`.
Manually adding `-fno-lto` would be another workaround.

`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream` as mentioned in https://github.com/espressif/arduino-esp32/issues/5436#issuecomment-885469600 to would also work.

VladislavSmolyanoy, 2022-02-22:
Found a solution that works on M1! 
https://github.com/espressif/arduino-esp32/issues/6269#issuecomment-1046149252

> 
> `platform.ini`:
> 
> ```ini
> extra_scripts =
> 	pre:disable_lto.py
> ```
> 
> `disable_lto.py`:
> 
> ```python
> Import("env")
> env.Append(
>   LINKFLAGS=[
>       "-fno-use-linker-plugin",
>   ]
> )
> ```

Jason2866, 2022-03-15:
Works without any issue with MacOS 12.3

Compiles without any errors or warnings with my custom Platformio setup for Tasmota (which can use lto).

Latest actual S3 branch Arduino build

```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/733/framework-arduinoespressif32-v4.4_dev-2509ea4227.tar.gz
```

llamaonaskateboard, 2022-03-30:
I can also confirm that everything is fine on the release MacOS 12.3 after reversing the workarounds.
Guess we can just assume the betas had a weird dylib behaviour.

valeros, 2022-04-18:
Closing the issue as everything works on the stable MacOS 12.3.

closed 2022-04-18 by valeros