platformio/platform-espressif32 #732: c_cpp_properties.json should not have comments

cyberman54, 2022-02-21:
The auto-generated file `c_cpp_properties.json` shows a comment at the top:
```
//
// !!! WARNING !!! AUTO-GENERATED FILE!
// PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
// https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
//
```
But JSON is data only. Thus, VScode complains _"comments not permitted in json"_ about.

ivankravets, 2022-02-21:
How to reproduce this issue? `c_cpp_properties.json` is not a generic JSON file, comments are allowed here. Do you use a custom JSON plugin that verifies files?

[image]

cyberman54, 2022-02-21:
Indeed it was a JSON plugin. Deinstalled it, now all fine. Sorry for wrong addressing this here. Closed.

closed 2022-02-21 by cyberman54