platformio/platform-espressif32 #735: CMakeLists.txt in test directory

ChrisEAlfred, 2022-03-01:
It is possible to have a CMakeLists.txt in the “test” directory?

I tried, but I don’t think the test/CMakeLists.txt was read from the test directory.

ChrisEAlfred, 2022-03-08:
Anyone?!

valeros, 2022-04-18:
Hi @ChrisEAlfred ! What do you plan to do in `test/CMakeLists.txt`?

ChrisEAlfred, 2022-04-18:
There are unit testing cases that require additional CMakeList.txt features:

- specifying the location of the components

`
set(EXTRA_COMPONENT_DIRS "../components")
`
- specifying test components

`
set(TEST_COMPONENTS "my_compoent1 my_component2" CACHE STRING "List of components to test")
`

valeros, 2022-04-19:
Unfortunately, it's not possible to use ESP-IDF's build system to unit test your code, mainly because PlatformIO has it's own [unit testing engine](https://docs.platformio.org/en/latest/plus/unit-testing.html) that may conflict with the Unity component within ESP-IDF.