platformio/platform-espressif32 #740: Unable to resolve dacDisable

MrYsLab, 2022-03-09:
I am using Ubuntu 21.04 and the latest version of VSCode. I have a file that compiles correctly using the Arduino IDE, but fails with platrormIO.  The board I selected is esp32doit-devkit-v1.

I have calls to both dacWrite and dacDisable. The dacWrite call resolves, but for some reason, dacDisble is not found. When trying to compile I get the following error:

```
error: 'dacDisable' was not declared in this scope
```

My main.cpp source code is:

```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  dacWrite(233);
  dacDisable(25)
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
My platform .ini is:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```

MrYsLab, 2022-03-09:
@Jason2866 Could you please tell me where to look to determine the version? I assumed that the latest version would be installed when I installed the extension.

I just tried modifying platform.ini with: platform = platformio/espressif32@^3.5.0.

Also, the example code above is not correct. It should be:
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  dacWrite(25, 233);
  dacDisable(25);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Jason2866, 2022-03-09:
If you use actual Arduino you use core 2.0.2. Official Platformio still uses core 1.0.6

MrYsLab, 2022-03-09:
It took a few hours searching on the web to find out how to update to the latest packages. I found[ this link](https://www.reddit.com/r/esp32/comments/rkpisx/how_can_i_use_arduinoesp32_version_201_or_version/) that solved the issue.

I changed platform.ini to:
```
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
board = esp32doit-devkit-v1
framework = arduino
```
I will close this issue and create a new issue on https://github.com/platformio/platformio-vscode-ide

closed 2022-03-09 by MrYsLab