platformio/platform-espressif32 #741: ESP32 (CH9102X) not auto-detected

moussie369, 2022-03-13:


FernandoGarcia, 2022-03-24:
Hi!

For me with Ubuntu 18.04 the board is detected and can use the serial monitor. Running `pio device list` I have:

```
/dev/ttyACM0
------------
Hardware ID: USB VID:PID=1A86:55D4 SER=5424032785 LOCATION=2-10:1.0
Description: USB Single Serial
```

In device list I have:

[image]

In windows I have:

[image]

For now I had no luck to compile a code for my `M5stack stamp C3` in PIO.

Best regards.

Jason2866, 2022-03-27:
@FernandoGarcia We can track down if this is a problem generated from the CH9102.
Trying this platform build. It supports for IDF/Arduino the ESP32, C3, S2 and S3
```
platform = https://github.com/Jason2866/platform-espressif32.git
```

closed 2022-09-20 by stale[bot]