platformio/platform-espressif32 #742: Support for Adafruit Feather ESP32-S2 TFT board

asciiphil, 2022-03-16:
The [Adafruit Feather ESP32-S2 TFT](https://www.adafruit.com/product/5300) does not appear to be available as a PlatformIO board.

The closest defined board is probably the Adafruit ESP32-S2 Feather Development Board, but the ESP32-S2 TFT Feather is distinct in at least a few ways.  At a minimum, the ESP32-S2 TFT Feather has 2MB PSRAM, compared to the 320KB RAM in the dev board's PlatformIO definition.

[Here's the board's entry in Espressif's boards.txt](https://github.com/espressif/arduino-esp32/blob/d977359e343bd1dfd83b82d14b6afc2a84fdd998/boards.txt#L5194).

asciiphil, 2022-03-20:
The following board definition works for me:

    {
        "build": {
            "arduino": {
                "ldscript": "esp32s2_out.ld"
            },
            "core": "esp32",
            "extra_flags": "-DBOARD_HAS_PSRAM",
            "f_cpu": "240000000L",
            "f_flash": "80000000L",
            "flash_mode": "dio",
            "hwids": [
                [ "0x239A", "0x810F" ],
                [ "0x239A", "0x010F" ],
                [ "0x239A", "0x8110" ]
            ],
            "mcu": "esp32s2",
            "variant": "adafruit_feather_esp32s2_tft"
        },
        "connectivity": [
            "wifi"
        ],
        "debug": {
            "openocd_target": "esp32s2.cfg"
        },
        "frameworks": [
            "arduino",
            "espidf"
        ],
        "name": "Adafruit ESP32-S2 TFT Feather",
        "upload": {
            "flash_size": "4MB",
            "maximum_ram_size": 327680,
            "maximum_size": 1310720,
            "require_upload_port": true,
            "speed": 460800
        },
        "url": "https://www.adafruit.com/product/5300",
        "vendor": "Adafruit"
    }

I haven't tested the debugging setting or the ESP-IDF platform.

Even though the boards.txt entry uses the qio flash mode, I couldn't get it to work with that and had to use dio.

The boards.txt has a menu for selecting USB CDC mode, MSC mode, or DFU mode at boot; I didn't see a way to do anything like that with PlatformIO, so I just left it out.  I just put the appropriate build flag in my `platformio.ini` file.  (`-DARDUINO_USB_CDC_ON_BOOT=1` in my case, since I want CDC mode.)

The stuff this needs from [arduino-esp32](https://github.com/espressif/arduino-esp32/) was only added in the 2.0.2 release, which isn't in PlatformIO yet (#619).  I had to pull it from git:

    [env:feather_esp32s2_tft]
    platform = 
    	https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages = 
    	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
    board = feather_esp32s2_tft
    framework = arduino

GitHubGoody, 2022-04-19:
I don't understand most of this, but I did just buy one of these boards hoping to integrate it with Home Assistant as my first ESPHome project. My understanding is that having it added to PlatformIO will enable it to work in ESPHome. Is that right? What's the next step? Thank you for working on this!

closed 2022-04-21 by valeros

valeros, 2022-04-25:
Hi @GitHubGoody , I have little knowledge of how ESPHome integrates with PlatformIO, but support for this board has been added to the development branch of this platform and will be released in the near future. It means you can use the upstream version of the platform without using the `platform_packages` option:
```
[env:feather_esp32s2_tft]
platform = https://github.com/platformio/platform-espressif32.git
board = feather_esp32s2_tft
framework = arduino
```

On a side note, it might be a good idea to delete the `~/.platformio` folder to avoid any possible issues with previously installed versions of the `espressif32` platform.

GitHubGoody, 2022-05-02:
@valeros Thank you. I wasn't able to get it working on ESPHome with those settings. But, I did just come across this:

https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32s2_tft.html#board-espressif32-adafruit-feather-esp32s2-tft

Is that more current than what you posted last week? As I attempt to translate to what ESPHome needs over on [that forum](https://community.home-assistant.io/t/first-esphome-project-esp32-s2-feather-board-stuck-on-initial-configuration-install/412097/36), I want to make sure I'm working off the right settings.

Specifically, I think the right settings would be:

```
#* [env:adafruit_feather_esp32s2_tft]
platform = espressif32
board = adafruit_feather_esp32s2_tft
framework = espidf
```
*I'M NOT SURE THIS IS NECESSARY FOR [WHAT ESPHOME PASSES TO PLATFORMIO.INI](https://esphome.io/components/esphome.html?highlight=platformio#esphome-platformio-options) SINCE THERE IS ONLY ONE ENV SECTION

Thanks again!

valeros, 2022-05-03:
Hi @GitHubGoody 

> Is that more current than what you posted last week?

Yes, that board is now available in the latest stable version of this platform [v4.2.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.2.0), so don't need the upstream platform version.

>  I wasn't able to get it working on ESPHome with those settings. 

I'm no expert on the ESPHome project, so it might be a good idea to ask questions directly in their repository.

commit 5f6a3f49 (2022-04-21): Add new boards DPU ESP32 and  Adafruit Feather ESP32-S2 TFT

Resolves #742