platformio/platform-espressif32 #746: Uart hang on overflow, fixed upstream

JBKingdon, 2022-03-26:
Hi,

There is a tricky bug in idf that causes uart input to stop if flush is called after a buffer overflow. It's been fixed in 4.4 and I was wondering if it could be back ported to 3.x while we're using that? I've also made a comment on the IDF issue to see if they are interested in back porting to 3.x.

IDF issue describing the problem (contains links to commits to fix the problem on 4.3 and 4.4):
https://github.com/espressif/esp-idf/issues/8445

barbudor, 2022-03-26:
Tasmota is running on ArduinoCore that is using IDF 4.4.0
So this should be part of Tasmota
@Jason2866 can confirm

Jason2866, 2022-03-26:
Outdated Arduino core 1.0.6 is based on IDF 3.3.
The actual Arduino Core 2.0.2 is based on IDF 4.4.
In a few days Arduino Core 2.0.3. will be released  based on more actual IDF 4.4.
Tasmota using its own Platformio Arduino Esp32 framework build which is based on latest IDF 4.4 and latest Arduino build branch S3 support.

In short when Platformio updates to actual Arduino core the issue is not there anymore.

JBKingdon, 2022-03-26:
This is all good, but platformIO supports native IDF and appears to currently be shipping 3.5. Perhaps I posted in the wrong place, but I'm hoping to have IDF fixed as well as the stacks built on top of it.

Jason2866, 2022-03-26:
> Hi,
> 
> There is a tricky bug in idf that causes uart input to stop if flush is called after a buffer overflow. It's been fixed in 4.4 and I was wondering if it could be back ported to 3.x while we're using that? I've also made a comment on the IDF issue to see if they are interested in back porting to 3.x.
> 
> IDF issue describing the problem (contains links to commits to fix the problem on 4.3 and 4.4): [espressif/esp-idf#8445](https://github.com/espressif/esp-idf/issues/8445)

Backporting is useless work since there is no support for Arduino core 1.0.x based on IDF3.3

Jason2866, 2022-03-26:
> This is all good, but platformIO supports native IDF and appears to currently be shipping 3.5. Perhaps I posted in the wrong place, but I'm hoping to have IDF fixed as well as the stacks built on top of it.

Yep, wrong place -> espressif idf github. Maybe it is already backported?
Than a new IDF 3.3 release is needed from espressif and Platformio has to update afterwards.
Probably this will not happen soon, since Platformio is a Ukraine project.

JBKingdon, 2022-03-26:
Ah, I see I've misunderstood the versioning as well. Thanks for the pointers. Feel free to close this issue if there's nothing to do on the platformIO side.

joba-1, 2022-03-26:
> if there's nothing to do on the platformIO side.

platformio usually follows releases automatically, but the process can be slow and is accelerated if there is a particular interest like yours.
Given that Espressif has mentioned that the 4.4 release containing the fix is imminent and there is a commit with the fix for 4.3, your issue may do the trick.
3.x is not mentioned. I would not expect a fix there because of its age.

> Tasmota using its own Platformio Arduino Esp32 framework build which is based on latest IDF 4.4 and latest Arduino build branch S3 support.

@Jason2866 : I would appreciate a heads up once the fix is in current tasmota esp32 builds. Then I could immediately test if it has an influence on c61 uart hangs.

JBKingdon, 2022-03-26:
Many thanks @joba-1 . Sorry for the confusion over 3.x, that was entirely my misunderstanding of what version I was using. I just checked .platformio/packages/framework-espidf/version.txt which shows 4.3.2 and makes much more sense.

barbudor, 2022-03-26:
My bad, I posted in platformio instead of Tasmota where Joba referenced this Issue
Please accept my apologies

Jason2866, 2022-03-26:
> platformio usually follows releases automatically, but the process can be slow and is accelerated if there is a particular interest like yours. Given that Espressif has mentioned that the 4.4 [...]

Our framework is based on latest IDF and Arduino. So the fix is there since a few weeks

Jason2866, 2022-03-29:
@JBKingdon You can use my updated fork. It supports IDF44 and Arduino 2.0.3-rc1 from today.
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
```

valeros, 2022-05-04:
Should be fixed in the latest version of the development platform that uses Arduino core v2.0.2, please update your packages via `pio update `. Feel free to reopen the issue if the problem persists.

closed 2022-05-04 by valeros

JBKingdon, 2022-05-04:
Many thanks for the update.