platformio/platform-espressif32 #748: undefined reference to `app_main'

praphullorahi, 2022-03-28:
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40302.0 (4.3.2)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
WARNING: You are using pip version 21.3.1; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\praphull.kumar\.platformio\penv\Scripts\python.exe -m pip install --upgrade pip' command.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TelldusLive> 0.1.0+sha.1dca1cf
Building in release mode
Linking .pio\build\bluesky\firmware.elf
c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\esp-idf\freertos\libfreertos.a(port_common.o):(.k+0x14): undefined reference to `app_main'
c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\esp-idf\freertos\libfreertos.a(port_common.o): i_task':
C:\Users\praphull.kumar\.platformio\packages\framework-espidf\components\freertos\port/port_common.c:129: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\bluesky\firmware.elf] Error 1

Jason2866, 2022-03-29:
Sketch which causes this error?

fedorenchik, 2022-03-30:
Common fix is to add `extern "C"` in front of `void app_main()`.

praphullorahi, 2022-03-30:
hello @fedorenchik 
I tried it before your comment , Added extern "C" before declaration of  void app_main()
 but still getting same error
 
 I would like to brief you more that app_main() is used in both main.c and port_common.c

fedorenchik, 2022-04-06:
@praphullorahi  You can also try `clean` or `cleanall` and then rebuild.

closed 2022-09-20 by stale[bot]