platformio/platform-espressif32 #749: getting linker error for espressif 3.5.0 with esp-idf 4.3.2

praphullorahi, 2022-03-28:
Linking .pio\build\bluesky\firmware.elf
c:/users/ultimate/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\src\provisioning.o:(.literal.provisioningStart+0x0): undefined reference to `wifi_prov_scheme_ble'
c:/users/ultimate/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\src\provisioning.o:(.literal.provisioningStart+0x4): undefined reference to `wifi_prov_scheme_ble_event_cb_free_btdm'
c:/users/ultimate/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\src\provisioning.o:(.literal.provisioningStart+0x34): undefined reference to `wifi_prov_scheme_ble_set_service_uuid'
c:/users/ultimate/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\bluesky\src\provisioning.o: in function `provisioningStart':
C:\Users\Ultimate\Desktop\bluesky-firmware/src/provisioning.c:122: undefined reference to `wifi_prov_scheme_ble_set_service_uuid'
collect2.exe: error: ld returned 1 exit status

Jason2866, 2022-03-29:
Without a sketch example, nobody knows what is causing the error.

AlphatestSylvain, 2022-05-03:
hi everyone,

Since I've update from 3.2.0 to the last release, I've exactly the same linled error.
And after several tests, I undersand that this issue appear with the 4.0.0 release.

Any idea ?

Thanks.

Jason2866, 2022-05-04:
@AlphatestSylvain  You need an updated Platformio setup which supports the changes in IDF 4.4.x.
It is NOT official released. See issues. You can try my adopted fork.
```
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-Tasmota-safe.zip
```

closed 2022-08-11 by stale[bot]