platformio/platform-espressif32 #750: Board Support: TTGO T-Energy ESP32 support

V1c99, 2022-03-30:
The TTGO T-Energy is an ESP32 WROVER-B based board, featuring a 18650 battery holder.

[T-Energy on LILYGO](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1170&FId=t3:50033:3)

[T-Energy on Amazon](https://www.amazon.com/TTGO-T-Energy-Bluetooth-ESP32-WROVER-B-Development/dp/B07KTVVPMJ)

Can you please add support for this board or send the required files to add it as a custom one

JRTax, 2022-12-16:
Is there a update on when this board is added?

peteDDD, 2023-12-12:
See this posting: [[Help with adding TTGO T-Energy ESP32 board](https://community.platformio.org/t/help-with-adding-ttgo-t-energy-esp32-board/26874)](https://community.platformio.org/t/help-with-adding-ttgo-t-energy-esp32-board/26874)