platformio/platform-espressif32 #753: Support cross-arch toolchain packages

quentinmit, 2022-04-01:


jpcornil-git, 2022-04-01:
It is a simple/working workaround when command line is used/available to end users (fake reported arch/force everything to 32b) but not when it is integrated in another environment like ESPHome (would require them to implement that workaround as well and undo it when proper support will be available to leverage 64b toolchains)

ivankravets, 2022-04-01:
@me-no-dev, could you publish to the registry the toolchains for patch2/patch3 for `linux_aarch64`?

```json
{
  "name": "toolchain-xtensa-esp32",
  "version": "8.4.0+2021r2-patch3",
  "description": "GCC Toolchain for Espressif ESP32 Xtensa MCUs",
  "keywords": [
    "build tools",
    "compiler",
    "assembler",
    "linker",
    "preprocessor",
    "espressif",
    "esp32"
  ],
  "license": "GPL-2.0-or-later",
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/crosstool-NG"
  },
  "system": [
    "linux_aarch64"
  ]
}
```

Thanks!

quentinmit, 2022-04-01:
@jpcornil-git I think your HAOS issue (32-bit userspace on 64-bit kernel) is really a bug in HAOS. They should be running the entire OS with the correct personality. (If you want to file an upstream bug, it looks like one way to do this is to set `Personality=arm` in (an override for) the systemd unit for docker.)

My issue here was for the feature request of supporting multiple architectures simultaneously; you only have one and pio is using the wrong one, which is a bug somewhere between HAOS and pio or both.

jpcornil-git, 2022-04-01:
@quentinmit Raspberry released 32b OS with both 32b and 64b kernels embedded for sometime (user-selectable cfr. [arm_64bit](https://www.raspberrypi.com/documentation/computers/config_txt.html#arm_64bit) ) while full 64b support is very recent cfr. [Raspberry Pi OS (64-bit)](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/).

On raspberry, ESPHome (the one using pio, not HAOS), may therefore run on 32b/32b, 32b/64b or 64b/64b OS/kernel platforms (and not always within a container). pio should therefore be robust against that in all cases and select the correct toochain based on the {machine, architecture} tuple.

Note that by controling what get_systype returns, e.g. a new platformio.ini param, you could also support multiple architecture simultaneously without relying on an external tool/wrapper and its limitations.

quentinmit, 2022-04-02:
Yeah, whether it's HAOS or Rasbian, if the whole userspace is 32-bit the kernel should be configured to run that userspace in the 32-bit personality (which causes `uname` / `platform.machine` to return a 32-bit arch). This isn't the only software that will be confused by the invalid architecture name.

jpcornil-git, 2022-04-02:
Reported kernel architecture is valid and software shouldn't be confused by that (apt, pip, ... package managers are not confused by kernel/userspace difference either). If get_systype() was returning **platform.(system,machine,architecture)** iso **platform.(system,machine)**, callers would have all required information to select correct toochain binaries., e.g. 
linux_aarch64_64bit, linux_aarch64_32bit, linux_armv7l_32bit for 64b/64b, 64b/32b, 32b/32b kernel/OS RPI platforms 

-> specifying/enforcing a personality is an an unnecessary overhead, information is available already

me-no-dev, 2022-04-02:
> @me-no-dev, could you publish to the registry the toolchains for patch2/patch3 for `linux_aarch64`?
>
> ```json
> {
>   "name": "toolchain-xtensa-esp32",
>   "version": "8.4.0+2021r2-patch3", [...]

I am uploading patch3 to the registry. Please use that and not patch2

ivankravets, 2022-06-14:
@me-no-dev , could you publish [linux-arm64.tar.gz](https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-linux-arm64.tar.gz) to the registry? The manifest is proposed above. Thanks!

The new report https://community.platformio.org/t/could-not-initialize-project/28325

me-no-dev, 2022-06-14:
@ivankravets uploading right now. Please figure out a way to version the core in such way that we can understand which OUR version is inside. What is 4.4.0?

me-no-dev, 2022-06-14:
@ivankravets 
```
pio package publish --non-interactive --owner espressif /Users/ficeto/Desktop/ESP32/pio_toolchain/toolchain-riscv32-esp-aarch64-8.4.0+2021r2-patch3.tar.gz
Preparing a package...
Type:     tool
Owner:    espressif
Name:     toolchain-riscv32-esp
System:   aarch64
Version:  8.4.0+2021r2-patch3
The package publishing may take some time depending on your Internet connection and the package size.
Publishing...
The package has been accepted. We will notify you on email when process it.
```

ivankravets, 2022-06-15:
> /Users/ficeto/Desktop/ESP32/pio_toolchain/toolchain-riscv32-esp-aarch64-8.4.0+2021r2-patch3.tar.gz

Could you open the `package.json` located in the archive? What is the value for the `system` field? The registry currently tracks 2 packages with non-existing systems:

- linux_aarch64linux_aarch64
- aarch64

The correct value is `linux_aarch64`. Could you fix the manifest and re-publish the package? We will remove redundant packages manually.

me-no-dev, 2022-06-16:
I deleted the packages, but will recreate them.

me-no-dev, 2022-06-16:
> The correct value is linux_aarch64. Could you fix the manifest and re-publish the package? We will remove redundant packages manually.

oops...

me-no-dev, 2022-06-16:
```json
  "name": "toolchain-riscv32-esp",
  "version": "8.4.0+2021r2-patch3",
  "description": "GCC Toolchain for Espressif 32-bit RISC-V based MCUs",
  "keywords": [
    "build tools",
    "compiler",
    "assembler",
    "linker",
    "preprocessor",
    "espressif",
    "esp32",
    "esp32c3",
    "risc-v"
  ],
  "license": "GPL-2.0-or-later",
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/crosstool-NG"
  },
  "system": "linux_aarch64"
}
```

me-no-dev, 2022-06-16:
https://registry.platformio.org/tools/espressif/toolchain-riscv32-esp/compatibility

ivankravets, 2022-06-16:
Wo-hoo! Finally fixed! Thanks! 🤗

closed 2022-06-16 by ivankravets

kaystrobach, 2022-10-08:
can we somehow for the usage of this version, so that we can workaround the problem from within the device yaml file?

related esphome issue: https://github.com/esphome/issues/issues/2722

Jason2866, 2022-10-08:
@kaystrobach One possibility use the PR https://github.com/platformio/platform-espressif32/pull/924 another using my fork with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip
```

kaystrobach, 2022-10-08:
thank you, in which file do i have to put this setting?

this does not look like the yaml syntax. thank you

kaystrobach, 2022-10-08:
I tried this in esphome config:

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    source: "https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip"
    platform_version: "2.0.5.1"
    version: 4.0.1
```

This causes an error like this:

```plain
Processing bhkw (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 2.0.5.1)
--------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1055, in main#
...
============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html
```

----

This installs something but fails because of an incompatibility:

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    #source: ""
    platform_version: "https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip"
    version: 4.0.1
```

```plain
Processing bhkw (board: esp32dev; framework: espidf; platform: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip
INFO Installing https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip
Unpacking  [####################################]  100%
Platform Manager: espressif32@2.0.5 has been installed!
INFO espressif32@2.0.5 has been installed!
Platform Manager: Removing espressif32 @ 2.0.5
INFO Removing espressif32 @ 2.0.5
Platform Manager: espressif32@2.0.5 has been removed!
INFO espressif32@2.0.5 has been removed!
Error: Development platform 'espressif32' is not compatible with PlatformIO Core v6.0.2 and depends on PlatformIO Core >=6.1.0.
```

Thank you for your support

Jason2866, 2022-10-08:
You have to upgrade platformio from your 6.0.2 version to 6.2 or greater.
If still errors occour i cant help any further how to integrate in esphome / home assistant. I do not use esphome nor Home Assistant.

kaystrobach, 2022-11-17:
@Jason2866 according to esphome devs a release of platformio is missing:

* https://github.com/esphome/issues/issues/2722#issuecomment-1317196966

Once this is fixed it should be solved for esphome as well - sry, but i do not understand the details here. Maybe you can do the release needed.

Jason2866, 2022-11-17:
@kaystrobach Not really the toolchains referred as missing **are** in Platformio since a few weeks. What is true Platformio did not do a Platform release which are **using** this toolchains.
In short HomeAssistant is waiting for a new Platform espressif32 release. All needed componenst and tools are already provided from espressif.

kaystrobach, 2022-11-17:
Ok, so now the HA guys need to do their *magic* to get it working in esphome? Or is the release still a thing? Is your fork online somewhere?

Jason2866, 2022-11-17:
Here it is. https://github.com/tasmota/platform-espressif32
How to use in github readme

danielbrunt57, 2022-12-01:
My head is spinning!
I've spent hours, no days, trying to follow all of this.
I did manage to install ESPHome on my PC, compile, flash and installed this device but I want to maintain it inside HA ESPHome on HA Blue ODROID.
Very simple on Windows x64.  I started with this in HA ESPHome on ODROID:
```
esphome:
  name: emporiavue2

external_components:
  - source: github://emporia-vue-local/esphome@emporia-vue-2022.4.0
    components: [ emporia_vue ]

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
```
```
INFO Reading configuration /config/esphome/emporiavue2.yaml...
INFO Detected timezone 'America/Vancouver'
INFO Generating C++ source...
INFO Compiling app...
Processing emporiavue2 (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
INFO Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```
```
and am now up to this:
```
esphome:
  name: emporiavue2
  platformio_options: 
    platform: espressif32
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
    platform_packages:
     - framework-espidf @ 3.40402.0
     - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
     - espressif/toolchain-esp32ulp @ ~2

external_components:
  - source: github://emporia-vue-local/esphome@emporia-vue-2022.4.0
    components: [ emporia_vue ]

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    source: "https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip"
    platform_version: "2.0.5.1"
    version: 4.0.1
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```
```
INFO Reading configuration /config/esphome/emporiavue2.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'America/Vancouver'
INFO Generating C++ source...
INFO Compiling app...
Processing emporiavue2 (board: esp32dev; framework: espidf; platform: espressif32)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
INFO Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'
```
Is there an end in sight, soon?

Jason2866, 2022-12-01:
> Is there an end in sight, soon?
I'm old (and currently very tired) but hopeful I see a solution, before I die!

Short version: No
Longer version: As long it is not possible to override **ALL** platform and platform_packages settings in EspHome it will not be possible. So either this happens or EspHome is updating there outdated ESP32 IDF build and toolchains.
Or, if you do not use any docker install variant of HA and EspHome you can change the relevant parts in source code to use the latest patch5 toolchains which are available for aarch64 here too. There are **no** aarch64 toolchains for patch2 and patch3 toolchains.

Jason2866, 2022-12-01:
@danielbrunt57 I did a quick hack of esphome to support Arduino/IDF with latest toolchain patch5.
Since i have no esp devices by hand (not at home), cant test if the successfully builded firmwares do work.
The fork is [here ](https://github.com/Jason2866/esphome)
Git clone the repo and install with `pip3 install -e .    `  when you are in the project folder

danielbrunt57, 2022-12-15:
@Jason2866 I think I did something wrong...
[image]

The existing directory is unchanged:
[image]

danielbrunt57, 2022-12-16:
After pondering on this for a while thinking about the desired outcome and since recovery from a botch-up is simple re & re of the HA add-on, I went ahead with a new strategy with `rm -r /Jason2866_esphome` as well as the original directory with `rm -r /esphome`.  Then a new try with `git clone <repo> /esphome; cd esphome; pip3 install -e` and opened up the HA's ESPHome.  All devices now needed a downgrade-update since there have been two new releases since your fork.  I tried emporiavue2 and it worked!
```
INFO Reading configuration /config/esphome/emporiavue2.yaml...
INFO Updating https://github.com/emporia-vue-local/esphome.git@emporia-vue-2022.4.0
INFO Detected timezone 'America/Vancouver'
INFO Generating C++ source...
INFO Compiling app...
Processing emporiavue2 (board: esp32dev; framework: espidf; platform: https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.5/platform-espressif32-v.2.0.5.zip)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.5/platform-espressif32-v.2.0.5.zip
INFO Installing https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.5/platform-espressif32-v.2.0.5.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2.0.5+20221125 has been installed!
INFO espressif32@2.0.5+20221125 has been installed!
<snip>
Linking /data/emporiavue2/.pioenvs/emporiavue2/bootloader.elf
Building /data/emporiavue2/.pioenvs/emporiavue2/bootloader.bin
Creating esp32 image...
Successfully created esp32 image.
Linking /data/emporiavue2/.pioenvs/emporiavue2/firmware.elf
RAM:   [=         ]   9.8% (used 32100 bytes from 327680 bytes)
Flash: [====      ]  40.8% (used 748753 bytes from 1835008 bytes)
Building /data/emporiavue2/.pioenvs/emporiavue2/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/emporiavue2/.pioenvs/emporiavue2/firmware.bin"], ["/data/emporiavue2/.pioenvs/emporiavue2/firmware.elf"])
Wrote 0xc8340 bytes to file /data/emporiavue2/.pioenvs/emporiavue2/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 349.10 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of emporiavue2.local
INFO  -> 192.168.1.10
INFO Uploading /data/emporiavue2/.pioenvs/emporiavue2/firmware.bin (754496 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from emporiavue2.local using esphome API
```

So, how do we convince Nabu Casa to upgrade ESPHome to support Arduino/IDF using the latest toolchain patch5?

danielbrunt57, 2022-12-16:
> esp32:
>   board: esp32doit-devkit-v1

Currently, the yaml file has:
```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
```
and I think the esp-idf is the issue?
Do you think the above change will work with `framework: type: esp-idf`??

danielbrunt57, 2022-12-16:
[image]

But then after adding:
```
  framework:
    type: esp-idf
```
[image]

[image]

Jason2866, 2022-12-16:
Ah, Esphome is still checking the toolchain version (in the new updated version where the latest platformio Platform is used). No dice.
You can have a look in my fork, what i have changed. Fork latest esphome and apply the changes i have done in my fork to your fork. So you can use latest esphome version.

danielbrunt57, 2022-12-16:
All this forking leads me back to...how to get esphome to fix the issue at hand?

danielbrunt57, 2022-12-16:
Ok, so, can you explain this???
I reverted to original yaml file and using latest esphome after previous attempt, attempted compile/installation, expecting it to fail, but it did not!
```
esphome:
  name: emporiavue2

external_components:
  - source: github://emporia-vue-local/esphome@emporia-vue-2022.4.0
    components: [ emporia_vue ]

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
    
# Enable logging
logger:
```
[image]

[logs_emporiavue2_upload (1).txt](https://github.com/platformio/platform-espressif32/files/10245190/logs_emporiavue2_upload.1.txt)

Jason2866, 2022-12-16:
Guessing, the check for the version of the installed toolchain is done only one time.

raoulbhatia, 2022-12-16:
FWIW, I stumbled across this issue as well for my `GL.iNet GL-S10` that I intend to use as a bluetooth-proxy, see [1][2][3].
I am running esphome in a docker container on a `Raspberry Pi 2 Model B Rev 1.1`.

It would be good to get that one fixed.
As a workaround, I can update from my laptop.  But this is not how I intend to use this product with esphome.

Thanks,
Raoul

[1] https://esphome.github.io/bluetooth-proxies/
[2] https://blakadder.com/gl-s10/
[3] https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml

Jason2866, 2022-12-16:
@danielbrunt57 Tell esphome crew to fix, everything needed is in platformio registry.
Platformio crew cant fix esphome stuff.

danielbrunt57, 2022-12-16:
Hi Jason,
I did some more analysis to try and figure out why this is working now vs all the failures before, but I can only surmise this:
When your forked esphome with patch5 failed, it must have downloaded something new into:
/config/esphome/.esphome/build/emporiavue2/src/esphome/components
/config/esphome/.esphome/build/emporiavue2/src/esphome/core
The /config/esphome folder is a shared folder in HA and is untouched by an add-on re & re.
Your forked esphome failed with `crosstool-ng version esp-2021r2-patch5 doesn't match supported version esp-2021r2-patch3` but after restoring HA's esphome, it's patch3 version was happy with what was in the shared folder.

I may attempt to clean out the shared folder (saving a copy first!) thus forcing esphome to start from scratch, and analyze it further to try and see what items are different, or can you put your finger on specific item(s) downloaded which HA esphome wouldn't do?

Jason2866, 2022-12-17:
@danielbrunt57  Updated my fork. Give it a try.

raoulbhatia, 2022-12-18:
@Jason2866 I am pretty new to all of this, so could you please explain why you are switching the `ARDUINO_PLATFORM_VERSION` from 5.2.0 to 2.0.5? 

```diff
-ARDUINO_PLATFORM_VERSION = cv.Version(5, 2, 0)
+ARDUINO_PLATFORM_VERSION = cv.Version(2, 0, 5)
```

FYI I am looking at https://github.com/Jason2866/esphome/pull/1

Jason2866, 2022-12-18:
@raoulbhatia it is my fork, i do not like the version scheme from platformio.
So i use a version scheme which is more logical to me. v2.0.5 is related to the Arduino core version.

danielbrunt57, 2022-12-19:
> Updated my fork. Give it a try.

Using this fork, ESPHome says "Update available" even after updating...

[image]

danielbrunt57, 2022-12-19:
I removed ESPHome (containing your forked version), removed /config/esphome/.esphome/build/emporiavue2, re-installed ESPHome and re-installed empriavue2.yaml and it installed fine all on its own now.  The original unsatisfied issue re: 'linux_aarch64' is no longer present.  IMO, what's required for this build is now available in native ESPHome...
```
INFO Reading configuration /config/esphome/emporiavue2.yaml...
INFO Detected timezone 'America/Vancouver'
INFO Generating C++ source...
INFO Compiling app...
Processing emporiavue2 (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
INFO Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-esp32ulp@2.35.0-20220830 has been installed!
INFO toolchain-esp32ulp@2.35.0-20220830 has been installed!
Tool Manager: Installing platformio/framework-espidf @ ~3.40402.0
INFO Installing platformio/framework-espidf @ ~3.40402.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.40402.0 has been installed!
INFO framework-espidf@3.40402.0 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40201.0 has been installed!
INFO tool-esptoolpy@1.40201.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.16.0
INFO Installing platformio/tool-cmake @ ~3.16.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-cmake@3.16.9 has been installed!
INFO tool-cmake@3.16.9 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
INFO Installing platformio/tool-ninja @ ^1.7.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-ninja@1.10.2 has been installed!
INFO tool-ninja@1.10.2 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40400.0
INFO Installing platformio/tool-scons @ ~4.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40400.0 has been installed!
INFO tool-scons@4.40400.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Installing ESP-IDF's Python dependencies
Collecting future>=0.15.2
  Downloading future-0.18.2.tar.gz (829 kB)
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting idf-component-manager~=1.0
  Downloading idf_component_manager-1.2.1-py3-none-any.whl (116 kB)
Collecting tqdm<5
[... 1254 lines cut ...]
Creating esp32 image...
Successfully created esp32 image.
Linking /data/emporiavue2/.pioenvs/emporiavue2/firmware.elf
RAM:   [=         ]   9.8% (used 32108 bytes from 327680 bytes)
Flash: [====      ]  40.4% (used 741541 bytes from 1835008 bytes)
Building /data/emporiavue2/.pioenvs/emporiavue2/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/emporiavue2/.pioenvs/emporiavue2/firmware.bin"], ["/data/emporiavue2/.pioenvs/emporiavue2/firmware.elf"])
Wrote 0xc6730 bytes to file /data/emporiavue2/.pioenvs/emporiavue2/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 363.73 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of emporiavue2.local
INFO  -> 192.168.1.10
INFO Uploading /data/emporiavue2/.pioenvs/emporiavue2/firmware.bin (747312 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from emporiavue2.local using esphome API
INFO Successfully connected to emporiavue2.local
[22:00:20][I][app:102]: ESPHome version 2022.12.1 compiled on Dec 18 2022, 21:57:19
[22:00:20][C][wifi:504]: WiFi:
```

Jason2866, 2022-12-19:
Great, just deleted my not needed anymore fork.