platformio/platform-espressif32 #755: ESP32-C3-MINI Serial monitor works in Arduino IDE but not with platform.io

ggalisky, 2022-04-04:
### Board

ESP32-C3-MINI Custom board

### Device Description

A ESP32-C3-MINI module with hand soldered wires and decoupling caps (see diagram below)

### Hardware Configuration

Hand wired module with below schematic

[image]

### Version

other

### IDE Name

Platform.io

### Operating System

Windows and Mac

### Flash frequency

80,000,000

### PSRAM enabled

no

### Upload speed

115200

### Description

Serial monitor is completely blank when uploading main.cpp when using Platform.io. The issue does not persist when using the Arduino IDE. The main.cpp runs fine after flashing, but I get no data on the serial monitor from platform.io

I believe it may have something to do with the "USB CDC" option. When I disable this in the Arduino IDE I get no serial output, when I enable it I get serial output. I've tried adding this to my platform.ini file but have not had any luck getting the serial monitor to work with PIO

### Sketch

```cpp
#include "Arduino.h"
void setup() {
    Serial.begin(115200);

}
void loop() {
    Serial.println("I'm alive");
    delay(100);
}
```
### ini
```
[env:arduino-esp32c3]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200

build_flags = 
    ; enable USB serial
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
    -D USBCON
```
### Debug Message

```plain
All I see on the serial monitor: 

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbmodem21401  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

### Other Steps to Reproduce

I've tried both MacOS and Windows and haven't been able to solve this. Been plugging away at it for the last few days. 

I've checked every thread I could find, but none have solved my issue. Thank you for your help

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-04:
Adding this build flags solves:
```
build_flags =
                -DARDUINO_USB_CDC_ON_BOOT=1
                -DARDUINO_USB_MODE=1
```
I have added boards.json in fork which has this settings set
https://github.com/Jason2866/platform-espressif32/blob/Arduino203rc1/boards/adafruit_qt_py_esp32-c3.json
https://github.com/Jason2866/platform-espressif32/blob/Arduino203rc1/boards/esp32-c3-mini.json

ggalisky, 2022-04-04:
@Jason2866 Thank you for responding to my issues an others quickly <3. **I appreciate you** and the work you are doing. This solved my issue.

closed 2022-04-04 by ggalisky

factoidforrest, 2024-01-22:
@Jason2866 These files are gone, have they gone somewhere else?

espressif32/blob/Arduino203rc1/boards/adafruit_qt_py_esp32-c3.json https://github.com/Jason2866/platform-espressif32/blob/Arduino203rc1/boards/esp32-c3-mini.json

Jason2866, 2024-01-22:
Should be here https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip

Do not recommend to use the old core 2.0.3. If you are just searching for the boards.json copy them from this build and use it in your project as user defined boards.