platformio/platform-espressif32 #757: build error: env windows:xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory

mirsys, 2022-04-07:
I use platformio espressif32 to compile the project https://github.com/sharandac/My-TTGO-Watch.git
An error occurred

Processing t-watch2020-v1 (platform: espressif32@3.5.0; board: ttgo-t-watch; framework: arduino)
-------------------------------------------------- -----------------------------------------Verbose mode can be enabled via `-v , --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (3.5.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny- h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016+sha.fb19175
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TTGO TWatch Library> 1.4.2
| |-- <Wire> 1.0.1
| |-- <SPI> 1.0
| |-- <Ticker> 1.1
| |-- <SPIFFS> 1.0
| | |-- <FS> 1.0
| |-- <FS> 1.0
| |-- <ESP32 BLE Arduino> 1.0.1
| |-- <SD(esp32)> 1.0.5
| | |-- <FS> 1.0
| | |-- <SPI> 1.0
|-- <lvgl> 7.7.2+sha.52f5239
| |-- <TTGO TWatch Library> 1.4.2
| | |-- <Wire> 1.0.1
| | |-- <SPI> 1.0
| | |-- <Ticker> 1.1
| | |-- <SPIFFS> 1.0
| | | |-- <FS> 1.0
| | |-- <FS> 1.0
| | |-- <ESP32 BLE Arduino> 1.0.1
| | |-- <SD(esp32)> 1.0.5
| | | |-- <FS> 1.0
| | | |-- <SPI> 1.0
| |-- <ESP32 BLE Arduino> 1.0.1
|-- <ESP Async WebServer> 1.2.3
| |-- <AsyncTCP> 1.1.1
| |-- <FS> 1.0
| |-- <WiFi> 1.0
| |-- <ArduinoJson> 6.19.4
|-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.19.4
|-- <ESP32SSDP> 1.2.0
| |-- <ESP32 Async UDP> 1.0.0
| |-- <WiFi> 1.0
|-- <IRremoteESP8266> 2.8.2
|-- <PubSubClient> 2.8.0
|-- <ESP8266Audio> 1.9.6
| |-- <FFat> 1.0
| | |-- <FS> 1.0
| |-- <FS> 1.0
| |-- <SPIFFS> 1.0
| | |-- <FS> 1.0
| |-- <HTTPClient> 1.2
| | |-- <WiFi> 1.0
| | |-- <WiFiClientSecure> 1.0
| | | |-- <WiFi> 1.0
| |-- <SD(esp32)> 1.0.5
| | |-- <FS> 1.0
| | |-- <SPI> 1.0
| |-- <SPI> 1.0
|-- <ESP8266SAM> 1.0.1
| |-- <ESP8266Audio> 1.9.6
| | |-- <FFat> 1.0
| | | |-- <FS> 1.0
| | |-- <FS> 1.0
| | |-- <SPIFFS> 1.0
| | | |-- <FS> 1.0
| | |-- <HTTPClient> 1.2
| | | |-- <WiFi> 1.0
| | | |-- <WiFiClientSecure> 1.0
| | | | |-- <WiFi> 1.0
| | |-- <SD(esp32)> 1.0.5
| | | |-- <FS> 1.0
| | | |-- <SPI> 1.0
| | |-- <SPI> 1.0
|-- <ESP8266FtpServer> 1.0.1
| |-- <FS> 1.0
| |-- <WiFi> 1.0
| |-- <SPIFFS> 1.0
| | |-- <FS> 1.0
|-- <ESP32-targz> 1.0.5-beta
| |-- <FS> 1.0
| |-- <Update> 1.0
| |-- <FFat> 1.0
| | |-- <FS> 1.0
| |-- <SD(esp32)> 1.0.5
| | |-- <FS> 1.0
| | |-- <SPI> 1.0
| |-- <SD_MMC> 1.0
| | |-- <FS> 1.0
| |-- <SPIFFS> 1.0
| | |-- <FS> 1.0
|-- <ESP Mail Client> 1.5.3+sha.f62eea3
| |-- <SPI> 1.0
| |-- <WiFi> 1.0
| |-- <SD_MMC> 1.0
| | |-- <FS> 1.0
| |-- <WiFiClientSecure> 1.0
| | |-- <WiFi> 1.0
| |-- <FS> 1.0
| |-- <SD(esp32)> 1.0.5
| | |-- <FS> 1.0
| | |-- <SPI> 1.0
| |-- <SPIFFS> 1.0
| | |-- <FS> 1.0
|-- <TinyGPSPlus> 1.0.3
|-- <EspSoftwareSerial> 6.15.2
|-- <FS> 1.0
|-- <SPIFFS> 1.0
| |-- <FS> 1.0
|-- <HTTPClient> 1.2
| |-- <WiFi> 1.0
| |-- <WiFiClientSecure> 1.0
| | |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <SD(esp32)> 1.0.5
| |-- <FS> 1.0
| |-- <SPI> 1.0
|-- <HTTPUpdate> 1.3
| |-- <HTTPClient> 1.2
| | |-- <WiFi> 1.0
| | |-- <WiFiClientSecure> 1.0
| | | |-- <WiFi> 1.0
| |-- <Update> 1.0
| |-- <WiFi> 1.0
|-- <Ticker> 1.1
|-- <ESP32 BLE Arduino> 1.0.1
|-- <twatch2021>
| |-- <Wire> 1.0.1
|-- <Wire> 1.0.1
|-- <Update> 1.0
Building in release mode
Linking .pio\build\t-watch2020-v1\firmware.elf
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\t-watch2020-v1\firmware.elf] Error 1
=============================== [FAILED] Took 21.24 seconds ============= ===================

PlatformIO.ini cfg

[env:t-watch2020-v1]
; overrides the default arduino-esp32 framework with an custom built arduino-esp32 framework
; the custom arduino-esp32 framework provides better power managment, dynamic frquency scaling and 80Mhz Flash/SPIRAM support
; built from https://github.com/espressif/esp32-arduino-lib-builder
platform_packages = framework-arduinoespressif32 @ https://github.com/sharandac/arduino-esp32-hedge.git
platform = espressif32@3.5.0
board = ttgo-t-watch
framework = arduino
lib_archive = true
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_speed = 115200
monitor_filters = 
	default
	esp32_exception_decoder
board_build.embed_txtfiles = 
	src/gui/mainbar/setup_tile/time_settings/timezones.json
	src/utils/osm_map/osmtileserver.json
src_filter = 
	+<*>
lib_deps = 
    TTGO TWatch Library@=1.4.2
    https://github.com/lvgl/lvgl.git#v7.7.2
    ESP Async WebServer@>=1.2.0
    AsyncTCP@>=1.1.1
    ArduinoJson@>=6.15.2
    luc-github/ESP32SSDP
    IRremoteESP8266@>=2.7.10
    PubSubClient@>=2.8
    earlephilhower/ESP8266Audio
    earlephilhower/ESP8266SAM
    nailbuster/ESP8266FtpServer
    https://github.com/tobozo/ESP32-targz/archive/refs/heads/1.0.5-beta.zip
    https://github.com/sharandac/ESP-Mail-Client
    mikalhart/TinyGPSPlus
    plerup/espsoftwareserial
build_flags = 
	-DCORE_DEBUG_LEVEL=3
    -D LILYGO_WATCH_2020_V1
	-mfix-esp32-psram-cache-issue
    -ffunction-sections
    -fdata-sections
    -Wl,--gc-sections
    -Os

closed 2022-04-08 by mirsys

kvatigabi, 2022-04-15:
What was the issue? I am experiencing the same thing.

Thermelgy-Repo, 2022-04-21:
@mirsys @kvatigabi I am also experiencing the same. Kindly guide us for solving it.

kvatigabi, 2022-04-21:
@Thermelgy-Repo I solved it by removing the `-Wl,--gc-sections` flags in `build_flags`. I left the other configuration the same.

Thermelgy-Repo, 2022-04-22:
@kvatigabi I am happy that you solved it. If you can give me exactly what to change that will be really helpful.

kvatigabi, 2022-04-22:
@Thermelgy-Repo I included it in my response before

> @Thermelgy-Repo I solved it by removing the `-Wl,--gc-sections` flags in `build_flags`. I left the other configuration the same.

Thermelgy-Repo, 2022-04-22:
@kvatigabi As you mentioned, -Wl, --gc-sections flags are not present in my Platform.ini file. 

````
[env:esp32dev]
platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#v3.5.0
board = esp32dev
framework = arduino

;platform_packages = 
; 	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

board_build.partitions = F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Partition\Thermelgy_Custom_16MB.csv

lib_extra_dirs = C:\Users\Asish Johney George\Documents\Arduino\libraries
monitor_speed = 115200
build_flags = 
	-L.pio/libdeps/esp32dev/mDash/src/esp32/ -llibmDash
	-DLOG_LEVEL=LOG_LEVEL_NONE
```

kvatigabi, 2022-04-23:
@Thermelgy-Repo  Are you using TTGO boards?
Your `board` is set to `esp32dev`

Thermelgy-Repo, 2022-04-25:
@kvatigabi I am using esp32 dev - WROOM32 module, not the TTGO.

Thermelgy-Repo, 2022-04-25:
I also solved this issue, this is happend because of incompactibiliy of versions of platform & platform_packages, Plz check that the Platform_packages version supports the Platform reselase of PlatformIO. Also refer to this issue link.

https://github.com/espressif/arduino-esp32/discussions/6611#discussion-4028123