platformio/platform-espressif32 #761: USB Host library missing

Spongman, 2022-04-18:
as described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_host.html

```
#include "usb/usb_host.h"
```

I created a new project:
```
[env:featheresp32-s2]
platform = espressif32
board = featheresp32-s2
framework = espidf
```

but `usb_host.h` is not found.

Spongman, 2022-04-18:
the whole `usb` directory is missing: https://github.com/espressif/esp-idf/tree/master/components/usb

valeros, 2022-04-18:
This component was only added in ESP-IDF v4.4. Unfortunately, there is no support for this version at the moment. The status is tracked here https://github.com/platformio/platform-espressif32/issues/710

closed 2022-04-18 by valeros