platformio/platform-espressif32 #763: Exception decoder for ESP32C3 is not working

kaizoku-oh, 2022-04-19:
I'm facing an issue with decoding ESP32C3 exceptions on serial monitor.
I also tried to add `build_type = debug` but no result.

This is my config:

```ini
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf

monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

Note that if I use ESP32 Dev the filter works fine:

```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

Are the ESP32**C3** exceptions decoding supported?

Jason2866, 2022-04-20:
Are you using latest `platform`develop version? Only there is the updated OCD v11 which supports the C3

Jason2866, 2022-04-26:
Not yet, there is a Draft PR #612

closed 2022-07-27 by kaizoku-oh