platformio/platform-espressif32 #769: Getting error while compiling after updating to V4.1.0

Thermelgy-Repo, 2022-04-22:
Dear team,

Thank you for the new release of platform-espressif32, As I was eagerly waiting for that. After updating it, I am getting error while building. It was compiling perfectly on 3.4 version. Kindly help me to solve it. 

**Error Details :** 
```
`c:/users/asish johney george/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/libdeps/esp32dev/mDash/src/esp32\libmDash.a:(.literal+0x534): undefined reference to `lwip_recvfrom_r'
c:/users/asish johney george/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/libdeps/esp32dev/mDash/src/esp32\libmDash.a:(.literal+0x548): undefined reference to `lwip_sendto_r'
c:/users/asish johney george/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/libdeps/esp32dev/mDash/src/esp32\libmDash.a: in function `mlog':
/Users/lsm/src/cesanta.com/alib/log.c:28: undefined reference to `lwip_recvfrom_r'
c:/users/asish johney george/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/libdeps/esp32dev/mDash/src/esp32\libmDash.a: in function `ll_read':
/Users/lsm/src/cesanta.com/alib/src/sock.c:102: undefined reference to `lwip_sendto_r'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`

```
**Platformio.ini File:**
`
```
[platformio]
default_envs = 

[env:esp32dev]
platform = espressif32
;platform = https://github.com/platformio/platform-espressif32.git#v3.5.0
board = esp32dev
framework = arduino
platform_packages = 
 	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
;	framework-arduinoespressif32@~2.0.2
;	toolchain-xtensa32@~2.80400.0
;	framework-arduinoespressif32@~2.0.2

;	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.2
board_build.partitions = F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Partition\Thermelgy_Custom_16MB.csv
lib_extra_dirs = C:\Users\Asish Johney George\Documents\Arduino\libraries
monitor_speed = 115200
build_flags = 
	-L.pio/libdeps/esp32dev/mDash/src/esp32/ -llibmDash
	-DLOG_LEVEL=LOG_LEVEL_NONE
lib_deps = 
	cesanta/mDash@^1.2.14
	eModbus@^1.3
	esphome/AsyncTCP-esphome@^1.2.2
	knolleary/PubSubClient@^2.8

[env:env1]
lib_deps = 
	eModbus@^1.3
	esphome/AsyncTCP-esphome@^1.2.2
	knolleary/PubSubClient@^2.8
`
```
**Build Details:**

```
Building...
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.3+sha.b3c203d (git+https://github.com/espressif/arduino-esp32.git)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 107 compatible libraries
Scanning dependencies...
```

valeros, 2022-04-22:
Hi @Thermelgy-Repo ! It seems that the `cesanta/mDash` library is not compatible with the Arduino core v2.0 based on ESP-IDF v4.4. You can either ask mDash developers to update the library [here](https://github.com/cesanta/mDash) or just use the version 3.5.0.

closed 2022-04-22 by valeros

Thermelgy-Repo, 2022-04-22:
Thank you so much @valeros