platformio/platform-espressif32 #773: Question: How to get `flash_extra_images` getting flashed?

Jason2866, 2022-04-25:
Board adafruit_funhouse_esp32s2.json has 
```
  "name": "Adafruit FunHouse",
  "upload": {
    "arduino": {
      "flash_extra_images": [
        [
          "0x2d0000",
          "variants/adafruit_funhouse_esp32s2/tinyuf2.bin"
        ]
      ]
    },
```
as entry, but this image is not beeing flashed.
Is there a additional setting in Platformio setup needed?

valeros, 2022-04-25:
Hi @Jason2866, The code responsible for processing these extra images is located in the Arduino build scripts.  I'm preparing a PR to the upstream Arduino with these new changes, so it will be available shortly. For now, you can download the `framework-arduinoespressif32` shipped with this platform and check the difference.

closed 2022-04-25 by valeros

Jason2866, 2022-04-25:
Thank you, now it does work :-)